Ulm-IQO/qudi-legacy #46: Any module relying on fitlogic will not load properly with lmfit-0.9.1

latchr, 2017-01-31:
When Qudi loads with default config, I had the following problem:

Used Qt API: PyQt4 (API v2)
Loading Qudi...
============= Starting Manager configuration from /home/lachlan/physics/code/qudi/config/example/default.cfg =================

============= Manager configuration complete =================

Remote connection not secured! Use a certificate!
Methods were included to FitLogic, but only if naming is right: check the doxygen documentation if you added a new method and it does not show.
TypeError
   <_ast.Module object at 0x7f4a5e06f358>
unsupported operand type(s) for *: 'float' and 'NoneType'
Error during activation
Traceback (most recent call last):
  File "/home/lachlan/physics/code/qudi/core/base.py", line 157, in _wrap_activation
    self.activate()
  File "/usr/lib64/python3.4/site-packages/fysom/__init__.py", line 300, in fn
    self.transition()
  File "/usr/lib64/python3.4/site-packages/fysom/__init__.py", line 295, in _tran
    self._after_event(e)
  File "/usr/lib64/python3.4/site-packages/fysom/__init__.py", line 326, in _after_event
    return getattr(self, fnname)(e)
  File "/usr/lib64/python3.4/site-packages/fysom/__init__.py", line 89, in _callback
    return func(obj, *args, **kwargs)
  File "/home/lachlan/physics/code/qudi/logic/optimizer_logic.py", line 166, in on_activate
    model, params = self._fit_logic.make_gausslinearoffset_model()
  File "/home/lachlan/physics/code/qudi/logic/fitmethods/gaussianlikemethods.py", line 153, in make_gausslinearoffset_model
    gauss_model, params = self.make_gaussoffset_model(prefix)
  File "/home/lachlan/physics/code/qudi/logic/fitmethods/gaussianlikemethods.py", line 127, in make_gaussoffset_model
    params = gauss_offset_model.make_params()
  File "/usr/lib64/python3.4/site-packages/lmfit/model.py", line 352, in make_params
    par = params[name] = Parameter(name=name)
  File "/usr/lib64/python3.4/site-packages/lmfit/parameter.py", line 72, in __setitem__
    self.update_constraints()
  File "/usr/lib64/python3.4/site-packages/lmfit/parameter.py", line 117, in update_constraints
    _update_param(name)
  File "/usr/lib64/python3.4/site-packages/lmfit/parameter.py", line 113, in _update_param
    self._asteval.symtable[name] = par.value
  File "/usr/lib64/python3.4/site-packages/lmfit/parameter.py", line 481, in value
    return self._getval()
  File "/usr/lib64/python3.4/site-packages/lmfit/parameter.py", line 454, in _getval
    check_ast_errors(self._expr_eval)
  File "/usr/lib64/python3.4/site-packages/lmfit/parameter.py", line 21, in check_ast_errors
    expr_eval.raise_exception(None)
  File "/usr/lib64/python3.4/site-packages/lmfit/asteval.py", line 146, in raise_exception
    raise exc(self.error_msg)
TypeError:  in expr='<_ast.Module object at 0x7f4a5e06f358>'
Task scannerLocationRefocus added!

latchr, 2017-01-31:
Upgrading lmfit from 0.9.1 to 0.9.5 has fixed this problem.  I don't know whether we should try to resolve the issue or simply mention somewhere in the documentation that Qudi doesn't work with 0.9.1?

a-stark, 2017-01-31:
Right, I had also encountered this issue and forgot to mention that in the pull request. We can mention that with one sentence in the docs, but the conda-env.yml has already the proper package number, so it should not concern further checkouts.

latchr, 2017-01-31:
Yes, but remember that some people will run Qudi without using the conda-env.  On my Gentoo Linux, for example, I have manually installed the required modules.  It is super easy to access lmfit.__version__ and I suggest that we put a small exception or error message to help anyone in this situation know what the easy fix is.  The original error message about TypeError is not very transparent at all!

a-stark, 2017-01-31:
That means to put in the activation of fitlogic a check for the version and issuing a warning or error. Sounds reasonable. FYI: I have lmfit.__version__ = '0.9.2' running on my windows machine and it works with the current fitlogic. So I assume from version '0.9.2' it works.

drogenlied, 2017-03-17:
Added an exception pointing out the version requirement in 3e61a5541a32fe4d9bbccdf87b04fa0122682814

closed 2017-03-17 by drogenlied