Ulm-IQO/qudi-legacy #47: Problem with pyqtgraph-0.9.10

latchr, 2017-01-31:
When trying to start Qudi I got the following problem:

Used Qt API: PyQt4 (API v2)
Loading Qudi...
============= Starting Manager configuration from /home/lachlan/physics/code/qudi/config/example/default.cfg =================

============= Manager configuration complete =================

Remote connection not secured! Use a certificate!
Methods were included to FitLogic, but only if naming is right: check the doxygen documentation if you added a new method and it does not show.
Task scannerLocationRefocus added!
No scanmode defined in config for odmr_logic module.
Falling back to list mode.
Error during activation
Traceback (most recent call last):
  File "/home/lachlan/physics/code/qudi/core/base.py", line 157, in _wrap_activation
    self.activate()
  File "/usr/lib64/python3.4/site-packages/fysom/__init__.py", line 300, in fn
    self.transition()
  File "/usr/lib64/python3.4/site-packages/fysom/__init__.py", line 295, in _tran
    self._after_event(e)
  File "/usr/lib64/python3.4/site-packages/fysom/__init__.py", line 326, in _after_event
    return getattr(self, fnname)(e)
  File "/usr/lib64/python3.4/site-packages/fysom/__init__.py", line 89, in _callback
    return func(obj, *args, **kwargs)
  File "/home/lachlan/physics/code/qudi/gui/odmr/odmrgui.py", line 166, in on_activate
    symbolSize=7
  File "/usr/lib64/python3.4/site-packages/pyqtgraph/graphicsItems/PlotDataItem.py", line 176, in __init__
    self.setData(*args, **kargs)
  File "/usr/lib64/python3.4/site-packages/pyqtgraph/graphicsItems/PlotDataItem.py", line 460, in setData
    self.updateItems()
  File "/usr/lib64/python3.4/site-packages/pyqtgraph/graphicsItems/PlotDataItem.py", line 492, in updateItems
    self.scatter.setData(x=x, y=y, **scatterArgs)
  File "/usr/lib64/python3.4/site-packages/pyqtgraph/graphicsItems/ScatterPlotItem.py", line 297, in setData
    self.addPoints(*args, **kargs)
  File "/usr/lib64/python3.4/site-packages/pyqtgraph/graphicsItems/ScatterPlotItem.py", line 398, in addPoints
    self.updateSpots(newData)
  File "/usr/lib64/python3.4/site-packages/pyqtgraph/graphicsItems/ScatterPlotItem.py", line 553, in updateSpots
    self.fragmentAtlas.getAtlas() # generate atlas so source widths are available.
  File "/usr/lib64/python3.4/site-packages/pyqtgraph/graphicsItems/ScatterPlotItem.py", line 193, in getAtlas
    self.buildAtlas()
  File "/usr/lib64/python3.4/site-packages/pyqtgraph/graphicsItems/ScatterPlotItem.py", line 186, in buildAtlas
    self.atlasData[x:x+w, y:y+h] = rendered[key]
TypeError: slice indices must be integers or None or have an __index__ method

latchr, 2017-01-31:
Upgrading from pyqtgraph-0.9.10 to pyqtgraph-0.10.0 made this problem go away.  Should we mention somewhere that Qudi seems incompatible with pyqtgraph-0.9.10?

a-stark, 2017-01-31:
That is similar to #46. I would suggest and prefer to update in general the [Required package list](https://ulm-iqo.github.io/qudi-generated-docs/html-docs/required-python-packages.html) which we have in the documentation, and stress that they have to be fulfilled.
Moreover I am not sure whether it is in general true that pyqtgraph-0.9.10 was not working with qudi, because this version exists since Dez. 2014 and you actually adapt the ScatterPlot object on 20.06.2016 to the current version. pyqtgraph-0.10.0 was released Nov. 2016.
Since pyqtgraph is used in almost all gui modules, it would be inadequate to check for the version in each gui module. I would also not include this check in the GUI base.

drogenlied, 2017-03-17:
pyqtraph 0.10.0 is required since we switched to Qt5 and is installed in the default environment.

We now have a conda environment for Linux, too, as in some distributions the included packages are several years old.

closed 2017-03-17 by drogenlied