Ulm-IQO/qudi-legacy #48: Linux install not working

crazy4pi314, 2017-02-01:
I have tried to install and launch Qudi on my Linux machine, but have run in to a problem. I installed the lists of dependencies in the tools folders (though this should be audited, do I really need SVN and devscripts to make Qudi work?). I generally prefer to use Anaconda distros of python but I went through and checked I had all of the packages listed installed.  However, when I try to run the start.py script I get the below error. 

**Kernel version:** Linux skaiser-XPS-15 4.4.0-59-generic #80-Ubuntu SMP Fri Jan 6 17:47:47 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

**Error message from start.py:**
```
skaiser@skaiser-XPS-15:~/git/qudi$ python start.py 
Used Qt API: PyQt4 (API v2)
Loading Qudi...
GLib-GIO-Message: Using the 'memory' GSettings backend.  Your settings will not be saved or shared with other applications.

(python:918): Gtk-WARNING **: GModule (/usr/lib/x86_64-linux-gnu/gtk-2.0/2.10.0/immodules/im-ibus.so) initialization check failed: GLib version too old (micro mismatch)

(python:918): Gtk-WARNING **: Loading IM context type 'ibus' failed

(python:918): Gtk-WARNING **: GModule (/usr/lib/x86_64-linux-gnu/gtk-2.0/2.10.0/immodules/im-ibus.so) initialization check failed: GLib version too old (micro mismatch)

(python:918): Gtk-WARNING **: Loading IM context type 'ibus' failed

(python:918): Gtk-WARNING **: GModule (/usr/lib/x86_64-linux-gnu/gtk-2.0/2.10.0/immodules/im-ibus.so) initialization check failed: GLib version too old (micro mismatch)

(python:918): Gtk-WARNING **: Loading IM context type 'ibus' failed
Traceback (most recent call last):
  File "/home/skaiser/anaconda3/lib/python3.5/runpy.py", line 184, in _run_module_as_main
    "__main__", mod_spec)
  File "/home/skaiser/anaconda3/lib/python3.5/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/skaiser/git/qudi/core/__main__.py", line 214, in <module>
    from .manager import Manager
  File "/home/skaiser/git/qudi/core/manager.py", line 36, in <module>
    from . import config
  File "/home/skaiser/git/qudi/core/config.py", line 38, in <module>
    import ruamel.yaml as yaml
ImportError: No module named 'ruamel'
Unexpected return value 1. Exiting.
```
I looked up this yaml loader and I do have the package that it comes from (PyYAML which was not a listed dependency). 

_nb: I have gotten this installed on my windows machine but I wont have many windows machines in the lab._

a-stark, 2017-02-01:
First, the svn dependency is really outdated and should be deleted from the list, thanks for mentioning. 
For a double check, what kind of yaml version do you have installed? I guess you have run the command 
`pip install pyyaml`.
You will need to have the ruamel.yaml package which should be obtained by
`pip install ruamel.yaml`.
Can you check whether this solves the issue?
To my knowledge, we did not test Qudi on Ubuntu machines, just on the debian jessie distribution.

Neverhorst, 2017-02-01:
I have created a ticket for updating the package list.
Btw, I have qudi running on Ubuntu (16.04 LTS).

crazy4pi314, 2017-02-01:
Installing ruamel.yaml worked, the update to the package listing (and maybe a python env please?) should fix it. Thanks!

drogenlied, 2017-04-25:
We now have a conda environment file and install script for Linux as well.

If anyone is interested in maintaining a package list for Debian or other Linux distributions they are welcome to contribute them.

It is unlikely that Qudi still works on Debian jessie, the Qt and Python requirements should necessitate at least Debain stretch. Therefore, the corresponding package lists will be removed for the next release.

closed 2017-04-25 by drogenlied