Ulm-IQO/qudi-legacy #114: Qudi crashes during restart while an IPython notebook was running with matplotlib

drogenlied, 2017-02-02:
Migration: trac#329.

 I have encountered this bug long long time ago and I can reproduce it very reliably. Qudi has a problem in restarting itself if it is connected to a jupyther notebook and a graph with matplotlib was plotted.

Here is how I can crash Qudi:

Open Qudi.

Run jupyter notebook in a shell
My run script for that look like (I know the script option is deprecated):

set mynotebookpath="C:\MyPath?"
cd "C:\Path-to-Anaconda-dir"
jupyter.exe notebook --notebook-dir=%mynotebookpath% --script

Start a new notebook and choose Qudi kernel.

Type in the following text in each line of the notebook and execute it:

# In[1]:

a = np.random.uniform(0,100,100)
b = np.linspace(0,100,100)

# In[2]:

import matplotlib.pyplot as plt

# In[3]:

plt.figure()
plt.plot(a,b)
plt.show()
plt.close()

Now restart Qudi. From here, every time I do that, I got a message:

Python has stopped working

Problem signature:

    Problem Event Name: BEX64
    Application Name: python.exe
    Application Version: 3.5.2150.1013
    Application Timestamp: 577be340
    Fault Module Name: ucrtbase.DLL
    Fault Module Version: 10.0.10240.16384
    Fault Module Timestamp: 559f3851
    Exception Offset: 0000000000064399
    Exception Code: c0000417
    Exception Data: 0000000000000000
    OS Version: 6.1.7601.2.1.0.256.4
    Locale ID: 1033
    Additional Information 1: 15e2
    Additional Information 2: 15e235c2234f6f4941bdaf904eccc57c
    Additional Information 3: b7f5
    Additional Information 4: b7f54178424fec095ce2276718152f80

Read our privacy statement online:

    ​http://go.microsoft.com/fwlink/?linkid=104288&clcid=0x0409

If the online privacy statement is not available, please read our privacy statement offline:

    C:\WINDOWS\system32\en-US\erofflps.txt

The log does not contain any bad stuff, in fact, it seems that Qudi quits normal. No error log was created. Here the log of the session:

2017-01-25 23:53:01 info main Loading Qudi...
2017-01-25 23:53:01 debug core.manager Main thread is <sip.voidptr object at 0x0000000003582CD8>
2017-01-25 23:53:01 info core.manager load.cfg config file found at C:\Users\astark\Dropbox\Doctorwork\Software\QuDi?-Git\qudi\config\load.cfg
2017-01-25 23:53:01 info core.manager Starting Manager configuration from C:/Users/astark/Dropbox/Doctorwork/Software/QuDi-Git/qudi/config/iqo-config/alex_pc/custom_alex.cfg
2017-01-25 23:53:01 info core.manager Manager configuration complete.
2017-01-25 23:53:01 info core.manager Remote port is the standard 12345
2017-01-25 23:53:01 debug core.threadmanager Creating thread: "rpyc-server".
2017-01-25 23:53:01 warning core.remote Remote connection not secured! Use a certificate!
2017-01-25 23:53:01 info core.remote Started module server at on port 12345
2017-01-25 23:53:01 info core.manager Qudi started.
2017-01-25 23:53:01 info core.manager Loading module ".gui.manager.managergui"
2017-01-25 23:53:01 info RemoteModule?/12345 server started on [0.0.0.0]:12345
2017-01-25 23:53:02 info core.manager Configuring ManagerGui? as man
2017-01-25 23:53:02 debug gui.manager.managergui.ManagerGui? Activation in thread <sip.voidptr object at 0x0000000003582CD8>
2017-01-25 23:53:04 debug gui.manager.managergui.ManagerGui? IPy activation in thread <sip.voidptr object at 0x0000000007CAFA08>
2017-01-25 23:53:04 info gui.manager.managergui.ManagerGui? IPython has kernel True
2017-01-25 23:53:04 info gui.manager.managergui.ManagerGui? IPython kernel alive True
2017-01-25 23:53:04 debug core.manager Activation success: True
2017-01-25 23:53:04 info core.manager Loading module ".gui.trayicon"
2017-01-25 23:53:04 info core.manager Configuring TrayIcon? as tray
2017-01-25 23:53:04 debug gui.trayicon.TrayIcon? Activation in thread <sip.voidptr object at 0x0000000005A29F58>
2017-01-25 23:53:04 debug core.manager Activation success: True
2017-01-25 23:53:04 info core.manager Loading module ".logic.taskrunner"
2017-01-25 23:53:04 info core.manager Configuring TaskRunner? as tasklogic
2017-01-25 23:53:04 debug core.threadmanager Creating thread: "mod-logic-tasklogic".
2017-01-25 23:53:04 debug logic.taskrunner.TaskRunner? Activation in thread <sip.voidptr object at 0x0000000005A29F30>
2017-01-25 23:53:04 info core.manager Task runner registered.
2017-01-25 23:53:04 debug core.manager Activation success: True
2017-01-25 23:53:20 info RemoteModule?/12345 accepted 127.0.0.1:60843
2017-01-25 23:53:20 info RemoteModule?/12345 welcome [127.0.0.1]:60843
2017-01-25 23:53:20 info core.remote Client connected!
2017-01-25 23:53:20 info core.remote remotesearch: kernellogic
2017-01-25 23:53:20 info core.remote remotesearch: kernellogic
2017-01-25 23:53:20 info core.manager Loading module ".logic.jupyterkernel.kernellogic"
2017-01-25 23:53:20 info core.manager Configuring QudiKernelLogic? as kernellogic
2017-01-25 23:53:20 info core.remote Shared module kernellogic.
2017-01-25 23:53:20 debug core.threadmanager Creating thread: "mod-logic-kernellogic".
2017-01-25 23:53:20 debug logic.jupyterkernel.kernellogic.QudiKernelLogic? Activation in thread <sip.voidptr object at 0x000000000838DBC0>
2017-01-25 23:53:21 debug core.manager Activation success: True
2017-01-25 23:53:21 info core.remote remoteload: logickernellogic
2017-01-25 23:53:21 info core.remote remotesearch: kernellogic
2017-01-25 23:53:21 info logic.jupyterkernel.kernellogic.QudiKernelLogic? Start {'hb_port': 56920, 'ip': '127.0.0.1', 'iopub_port': 56917, 'key': '045100f3-f57c-4f34-a147-509354901f19', 'stdin_port': 56918, 'signature_scheme': 'hmac-sha256', 'transport': 'tcp', 'control_port': 56919, 'shell_port': 56916}
2017-01-25 23:53:21 info root New Kernel ca51918b-bca0-4c52-96c1-6fe544deed7b
2017-01-25 23:53:21 info logic.jupyterkernel.kernellogic.QudiKernelLogic? Kernel is ca51918b-bca0-4c52-96c1-6fe544deed7b
2017-01-25 23:53:21 info logic.jupyterkernel.kernellogic.QudiKernelLogic? Finished starting Kernel ca51918b-bca0-4c52-96c1-6fe544deed7b
2017-01-25 23:53:21 info root Config: {"hb_port": 56920, "ip": "127.0.0.1", "iopub_port": 56917, "key": "045100f3-f57c-4f34-a147-509354901f19", "stdin_port": 56918, "signature_scheme": "hmac-sha256", "transport": "tcp", "control_port": 56919, "shell_port": 56916}
2017-01-25 23:53:21 info root Starting loops...
2017-01-25 23:53:21 info root Ready! Listening...
2017-01-25 23:53:30 info core.manager load.cfg config file found at C:\Users\astark\Dropbox\Doctorwork\Software\QuDi?-Git\qudi\config\load.cfg
2017-01-25 23:53:30 info core.manager Set loaded configuration to C:/Users/astark/Dropbox/Doctorwork/Software/QuDi-Git/qudi/config/iqo-config/alex_pc/custom_alex.cfg
2017-01-25 23:53:30 info core.manager Restarting Qudi after configuration reload.
2017-01-25 23:53:30 info core.manager Deactivating gui.man
2017-01-25 23:53:30 debug gui.manager.managergui.ManagerGui? Deactivation in thread <sip.voidptr object at 0x000000000A4F7CD8>
2017-01-25 23:53:30 debug gui.manager.managergui.ManagerGui? IPy deactivation: <sip.voidptr object at 0x000000000A4F7CD8>
2017-01-25 23:53:30 debug core.manager Deactivation success: True
2017-01-25 23:53:30 info core.manager Deactivating gui.tray
2017-01-25 23:53:30 debug gui.trayicon.TrayIcon? Deactivation in thread <sip.voidptr object at 0x000000000AFC8DF0>
2017-01-25 23:53:30 debug core.manager Deactivation success: True
2017-01-25 23:53:30 info core.manager Deactivating logic.tasklogic
2017-01-25 23:53:30 debug logic.taskrunner.TaskRunner? Deactivation in thread <sip.voidptr object at 0x000000000AFC8DF0>
2017-01-25 23:53:30 info core.manager Task runner removed.
2017-01-25 23:53:30 debug core.threadmanager Quitting thread mod-logic-tasklogic.
2017-01-25 23:53:30 debug core.threadmanager Waiting for thread mod-logic-tasklogic to end.
2017-01-25 23:53:30 debug core.manager Deactivation success: True
2017-01-25 23:53:30 debug core.threadmanager Thread mod-logic-tasklogic has quit.
2017-01-25 23:53:30 debug core.threadmanager Cleaning up thread mod-logic-tasklogic.
2017-01-25 23:53:30 info core.manager Deactivating logic.kernellogic
2017-01-25 23:53:30 debug logic.jupyterkernel.kernellogic.QudiKernelLogic? Deactivation in thread <sip.voidptr object at 0x000000000AFC8DF0>
2017-01-25 23:53:30 info logic.jupyterkernel.kernellogic.QudiKernelLogic? Stopping ca51918b-bca0-4c52-96c1-6fe544deed7b
2017-01-25 23:53:30 info logic.jupyterkernel.kernellogic.QudiKernelLogic? Cleanup kernel ca51918b-bca0-4c52-96c1-6fe544deed7b
2017-01-25 23:53:30 debug core.threadmanager Quitting thread mod-logic-kernellogic.
2017-01-25 23:53:30 debug core.threadmanager Waiting for thread mod-logic-kernellogic to end.
2017-01-25 23:53:30 debug core.manager Deactivation success: True
2017-01-25 23:53:30 info main Closing windows...
2017-01-25 23:53:30 info main Stopping threads...
2017-01-25 23:53:30 debug core.threadmanager Quit all threads.
2017-01-25 23:53:30 info main Qudi is closed! Ciao.

drogenlied, 2017-03-27:
I found out that this seems to happen only on Windows. Linux native and Linux conda environments do not crash in this scenario.

a-stark, 2017-08-23:
It seems that this crash happens also in AppVeyor and TravisCI, but it is not recognized as a crash! The following build logs are mainly taken from the merge of #250.

For the AppVeyor, just compare [this](https://ci.appveyor.com/project/InstituteforQuantumOptics/qudi/build/0.6.0.822) _successful_ build with [this](https://ci.appveyor.com/project/InstituteforQuantumOptics/qudi/build/0.6.0.821)  _failed_ build. The returned message is always the same:

```
Unexpected return value 3221226505. Exiting.
C:\projects\qudi\tools\test.sh: line 66: kill: (3012) - No such process
```

Also the TracisCI [unix based](https://travis-ci.org/Ulm-IQO/qudi/builds/267097510?utm_source=github_status&utm_medium=notification) test has a problem during the shutdown:

```
QSocketNotifier: Invalid socket 17 and type 'Read', disabling...
./tools/test.sh: line 66: kill: (8913) - No such process
```

Under Windows 7, I obtain a similar error message from the cpython interpreter like in the AppVeyor build above:

`Unexpected return value 3221226519. Exiting.`

Does someone know from where I can get the meanings of those return values?

a-stark, 2017-08-23:
Maybe another interesting hint. I executed the code above in the debug mode of PyCharm. Here I obtain an error in the brower, which was not displayed before. When the part
`plt.plot(a,b)`
is executed I obtain the following traceback (which seems to be ignored):

```
Exception ignored in: <generator object _process_plot_var_args._getdefaults.<locals>.<genexpr> at 0x000000000DFBB570>
Traceback (most recent call last):
  File "C:\Anaconda3_64\lib\site-packages\matplotlib\axes\_base.py", line 275, in <genexpr>
    for k in prop_keys):
SystemError: error return without exception set
Exception ignored in: <generator object Transform._iter_break_from_left_to_right at 0x000000000DFBB8E0>
Traceback (most recent call last):
  File "C:\Anaconda3_64\lib\site-packages\matplotlib\transforms.py", line 1210, in _iter_break_from_left_to_right
    yield IdentityTransform(), self
SystemError: error return without exception set
Exception ignored in: <generator object CompositeGenericTransform._iter_break_from_left_to_right at 0x000000000DFBB888>
Traceback (most recent call last):
  File "C:\Anaconda3_64\lib\site-packages\matplotlib\transforms.py", line 2345, in _iter_break_from_left_to_right
    yield lh_compliment, rh_compliment + self._b
SystemError: error return without exception set
```
@drogenlied or @tobiasgehring do you have any idea about that?

drogenlied, 2017-08-24:
The return values that you see in Windows are the NTSTATUS, see https://msdn.microsoft.com/en-us/library/cc704588.aspx

a-stark, 2017-08-24:
Thanks a lot @drogenlied!
So the error message I get is then

```
STATUS_INVALID_CRUNTIME_PARAMETER
0xC0000417 (=3221226519)
An invalid parameter was passed to a C runtime function.
```
The return values from the AppVeyor Scripts correspond to

```
STATUS_STACK_BUFFER_OVERRUN
0xC0000409  (=3221226505)
The system detected an overrun of a stack-based buffer in this application. 
This overrun could potentially allow a malicious user to gain control of this application.
```

Neverhorst, 2017-09-12:
Just wanted to report that I have (almost) the same issue on all my windows machines running qudi. (Win7 64bit Enterprise, Win8.1 64bit Enterprise)

Upon exiting qudi (not only restart but also just quitting) I get the STATUS_STACK_BUFFER_OVERRUN error as described above with the problem signature as initially posted. 

This happens 100% of times I save a figure using matplotlib via savelogic despite properly closing the figure afterwards.

closed 2018-02-26 by a-stark