Ulm-IQO/qudi-legacy #49: Offset-anchor optimization technique in POI Manger

drogenlied, 2017-02-02:
Migration: trac#3.

It should be possible to re-optimize the alignment on a chosen POI by algorithmically going to a different POI and finding the current sample position. This would allow a POI to be tracked even when it is in a local fluorescence environment that is not suitable for direct optimization.

Neverhorst, 2018-11-22:
I think this will be resolved by PR #366, right @kay-jahnke?

kay-jahnke, 2018-11-23:
Yes, PR #366 resolves this one. I do however need to finish the PR first.
Gaussian fit onto the convolution in XY direction is still missing and turned our to be crucial.

liuyanulm, 2020-10-07:
I believe this issue has been resolved.

Neverhorst, 2020-10-07:
Actually this has not been resolved since PR #366 has been withdrawn without merging.