Ulm-IQO/qudi-legacy #56: Last spin box/text field change is not applied when clicking a button

drogenlied, 2017-02-02:
Migration: trac#34.

Example:
Starting a scan triggers editingFinished for input fields after processing the signal from the start button and makes the scan with the old values.

drogenlied, 2017-02-02:
It is possible to fix this by using valueChanged instead of editingFiniahed, but that causes errors when entries are incomplete. For example, divorce would complain if you typed a "-" character before any numbers. However, since we now use spin boxes they are giving us implicit input parsing and perhaps we could go back to valueChanged.

liuyanulm, 2020-10-07:
This is been resolved.

Neverhorst, 2020-10-07:
This is a general problem of almost any GUI in qudi. We should only close this PR if we have solved this issue in a general way by some workaround or Qt changing its behaviour (unlikely).