Ulm-IQO/qudi-legacy #57: Get Confocal Image does not get the image bounds correctly the first time

drogenlied, 2017-02-02:
Migration: trac#42.

Steps to reproduce:
 * Make confocal scan.
 * Open PoiManager, the confocal image is loaded automatically.
 * Make a new scan with different ranges.
 * Click Get Confocal Image. The image data updates, but the position does not.
 * Click Get Confocal Image again. Now the position of the scan is updated correctly.

kay-jahnke, 2017-02-14:
@drogenlied : I tried with dummy configuration and a clean App-Status to reproduce the bug.
I can't find anything wrong with the POI-Manager. Everything works for me and the axis gets updated to the new range.

The only thing I noticed was that in confocal, after changing the scan-region, the crosshair was outside the scan, but the picture scaled nicely, so that was OK.

drogenlied, 2017-02-14:
I cannot reproduce this on a real setup either, it must have been fixed by other changes.
Feel free to close this issue.

kay-jahnke, 2017-02-14:
Issues seems to have been resolved sometime in the past.

closed 2017-02-14 by kay-jahnke