Ulm-IQO/qudi-legacy #58: There is no documentation to show what is available for inclusion in the config file

drogenlied, 2017-02-02:
Migration: trac#44.

There is no easy way to know what is available for inclusion in the config file. Up until now the default.cfg has been a sort of "catch-all" exhaustive list, but this makes less and less sense as we get more real hardware modules.

drogenlied, 2017-02-02:
This is not quite true. In the tools directory there is a listmods.py tool which shows the available modules for including in config. However, this process should still be documented in a way that makes it easier for an "outsider" to step through setting up QuDi? on their own experiment.

liuyanulm, 2020-10-07:
Documentation has been implemented. closing this issue.

Neverhorst, 2020-10-07:
> This is not quite true. In the tools directory there is a listmods.py tool which shows the available modules for including in config. However, this process should still be documented in a way that makes it easier for an "outsider" to step through setting up QuDi? on their own experiment.

There is still no documentation about where to find all available modules and how to include them in the config.

commit f88e762d (2016-05-10): fixed #58 by passing proper arguments to signal_refocus_finished if the refocus is aborted, and some PEP8 compliance in optimizer_logic