Ulm-IQO/qudi-legacy #59: Crash in confocal while changing color bar

drogenlied, 2017-02-02:
Migration: trac#45.

 Fatal Python error: Segmentation fault

    Thread 0x000292fc (most recent call first):

    File "C:\Anaconda3\lib\site-packages\PyDAQmx\DAQmxFunctions.py", line 24 in mafunction
    File "<string>", line 2 in function
    File "C:\software\QuDi?\hardware\ni_card.py", line 1008 in _write_scanner_ao
    File "C:\software\QuDi?\hardware\ni_card.py", line 1197 in scan_line
    File "C:\software\QuDi?\logic\confocal_logic.py", line 479 in _scan_line

    Thread 0x0002928c (most recent call first):

    File "C:\Anaconda3\lib\site-packages\PyDAQmx\DAQmxFunctions.py", line 24 in mafunction
    File "<string>", line 2 in function
    File "C:\software\QuDi?\hardware\ni_card.py", line 607 in get_counter
    File "C:\software\QuDi?\logic\counter_logic.py", line 363 in countLoopBody_continuous

    Thread 0x00029244 (most recent call first):

    File "C:\Anaconda3\lib\site-packages\zmq\utils\garbage.py", line 47 in run
    File "C:\Anaconda3\lib\threading.py", line 920 in _bootstrap_inner
    File "C:\Anaconda3\lib\threading.py", line 888 in _bootstrap

    Thread 0x00029224 (most recent call first):
  
    File "C:\Anaconda3\lib\threading.py", line 290 in wait
    File "C:\Anaconda3\lib\threading.py", line 553 in wait
    File "C:\Anaconda3\lib\site-packages\IPython\core\history.py", line 791 in run
    File "C:\Anaconda3\lib\site-packages\IPython\core\history.py", line 68 in needs_sqlite
    File "<string>", line 2 in run
    File "C:\Anaconda3\lib\threading.py", line 920 in _bootstrap_inner
    File "C:\Anaconda3\lib\threading.py", line 888 in _bootstrap

    Thread 0x00029204 (most recent call first):

    File "C:\Anaconda3\lib\socket.py", line 187 in accept
    File "C:\Anaconda3\lib\site-packages\rpyc\utils\server.py", line 128 in accept
    File "C:\Anaconda3\lib\site-packages\rpyc\utils\server.py", line 241 in start
    File "C:\software\QuDi?\core\remote.py", line 168 in run

    Current thread 0x000291fc (most recent call first):

    File "C:\software\QuDi?\gui\confocal\confocalgui.py", line 692 in get_xy_cb_range
    File "C:\software\QuDi?\gui\confocal\confocalgui.py", line 1418 in refresh_xy_image
    File "C:\software\QuDi?\gui\confocal\confocalgui.py", line 1360 in update_xy_cb_range
    File "C:\software\QuDi?\gui\confocal\confocalgui.py", line 1343 in shortcut_to_xy_cb_centiles
    File "C:\software\QuDi?\core\main.py", line 198 in <module>
    File "C:\Anaconda3\lib\runpy.py", line 85 in _run_code
    File "C:\Anaconda3\lib\runpy.py", line 170 in _run_module_as_main

drogenlied, 2017-02-02:
Hi guys,
in my opinion, the error messages are not telling really what has happened and which causes the Segmentation Fault. I would say that their crash is just the result of something, which happens before (which is not caught by the Traceback).
I can reproduce the error 'relatively reliable' by pressing the confocal to scan and changing with the mouse wheel at first the upper centiles and directly afterwards moving with the mouse wheel the lower centiles back and forth.
Moreover, I do not think that we would be able to trace that error back to its origin, I believe that is a very tricky bug either in in Qt or even in the c-code of the python interpreter. I would guess it has to do with the dereferencing of a value passed by a signal of Qt.
But you can obviously see the problem here: The burst of signals resulting from changing the spinbox will eventually change the colorbar itself and the picture, which have cross dependencies during the signal sent process. So my suggestion is actually quite simple: Use a SignalProxy? to reduce the burst of emitted signals during the scan (I have added the file QuDi?-dev\Playground\proxy-for-signals.py some time ago exactly for that purpose). I believe that would solve the problem. I.e. deconnect the change of the spinbox upon start of the scan and connect before scan actually starts the value change via the SignalProxy? to reduce the signals (you will actually not feel the difference if signals are reduced to max every 50ms or higher). If the scan stops or is finished, deconnect the Proxy and connect the normal changed method.
Or make it simple and connect the proxy the whole time with the value changed signal.

;)

If you have time to do it you are free to go, I would only have time for that after the pulsed environment works properly.

drogenlied, 2017-02-02:
Up to date backtrace:

    Fatal Python error: Segmentation fault

    Thread 0x000016c4 (most recent call first):

    File "C:\Anaconda3\envs\qudi\lib\site-packages\PyDAQmx\DAQmxFunctions.py", line 25 in mafunction
    File "<string>", line 2 in function
    File "C:\software\QuDi?\hardware\ni_card.py", line 1160 in _write_scanner_ao
    File "C:\software\QuDi?\hardware\ni_card.py", line 1345 in scan_line
    File "C:\software\QuDi?\logic\confocal_logic.py", line 664 in _scan_line

   Thread 0x00001c14 (most recent call first):

    File "C:\software\QuDi?\core\parentpoller.py", line 105 in run
    File "C:\Anaconda3\envs\qudi\lib\threading.py", line 920 in _bootstrap_inner
    File "C:\Anaconda3\envs\qudi\lib\threading.py", line 888 in _bootstrap

   Thread 0x000014b0 (most recent call first):

    File "C:\Anaconda3\envs\qudi\lib\site-packages\zmq\utils\garbage.py", line 47 in run
    File "C:\Anaconda3\envs\qudi\lib\threading.py", line 920 in _bootstrap_inner
    File "C:\Anaconda3\envs\qudi\lib\threading.py", line 888 in _bootstrap

    Thread 0x00001ddc (most recent call first):

    File "C:\Anaconda3\envs\qudi\lib\threading.py", line 290 in wait
    File "C:\Anaconda3\envs\qudi\lib\threading.py", line 553 in wait
    File "C:\Anaconda3\envs\qudi\lib\site-packages\IPython\core\history.py", line 798 in run
    File "C:\Anaconda3\envs\qudi\lib\site-packages\IPython\core\history.py", line 68 in needs_sqlite
    File "<decorator-gen-23>", line 2 in run
    File "C:\Anaconda3\envs\qudi\lib\threading.py", line 920 in _bootstrap_inner
    File "C:\Anaconda3\envs\qudi\lib\threading.py", line 888 in _bootstrap

    Thread 0x00001bd0 (most recent call first):

    File "C:\Anaconda3\envs\qudi\lib\socket.py", line 187 in accept
    File "C:\Anaconda3\envs\qudi\lib\site-packages\rpyc\utils\server.py", line 128 in accept
    File "C:\Anaconda3\envs\qudi\lib\site-packages\rpyc\utils\server.py", line 241 in start
    File "C:\software\QuDi?\core\remote.py", line 184 in run

    Current thread 0x00000d78 (most recent call first):

    File "C:\software\QuDi?\gui\confocal\confocalgui.py", line 735 in get_xy_cb_range
    File "C:\software\QuDi?\gui\confocal\confocalgui.py", line 778 in refresh_xy_colorbar
    File "C:\software\QuDi?\gui\confocal\confocalgui.py", line 1472 in refresh_xy_image
    File "C:\software\QuDi?\gui\confocal\confocalgui.py", line 1410 in update_xy_cb_range
    File "C:\software\QuDi?\gui\confocal\confocalgui.py", line 1393 in shortcut_to_xy_cb_centiles
    File "C:\software\QuDi?\core\main.py", line 233 in <module>
    File "C:\Anaconda3\envs\qudi\lib\runpy.py", line 85 in _run_code
    File "C:\Anaconda3\envs\qudi\lib\runpy.py", line 170 in _run_module_as_main

   The offending line is:

    # Exclude any zeros (which are typically due to unfinished scan)
    xy_image_nonzero = self.xy_image.image[ np.nonzero(self.xy_image.image) ]

Neverhorst, 2017-03-17:
Confirmed. 
Also happens randomly at my setup.

Neverhorst, 2017-05-02:
Up to date error that occured while scanning and changing the upper percentile spinbox with the mouse wheel:

Windows fatal exception: access violation

Thread 0x000001d0 (most recent call first):
  File "C:\Anaconda3\envs\qudi\lib\site-packages\PyDAQmx\DAQmxFunctions.py", line 25 in mafunction
  File "<string>", line 2 in function
  File "C:\Software\qudi\hardware\ni_card.py", line 1264 in _write_scanner_ao
  File "C:\Software\qudi\hardware\ni_card.py", line 1432 in scan_line
  File "C:\Software\qudi\logic\interfuse\scanner_tilt_interfuse.py", line 188 in scan_line
  File "C:\Software\qudi\logic\confocal_logic.py", line 744 in _scan_line

Thread 0x00001cec (most recent call first):
  File "C:\Anaconda3\envs\qudi\lib\site-packages\PyDAQmx\DAQmxFunctions.py", line 25 in mafunction
  File "<string>", line 2 in function
  File "C:\Software\qudi\hardware\ni_card.py", line 773 in get_counter
  File "C:\Software\qudi\logic\counter_logic.py", line 477 in count_loop_body

Thread 0x00001c6c (most recent call first):
  File "C:\Software\qudi\core\parentpoller.py", line 132 in run
  File "C:\Anaconda3\envs\qudi\lib\threading.py", line 916 in _bootstrap_inner
  File "C:\Anaconda3\envs\qudi\lib\threading.py", line 884 in _bootstrap

Thread 0x00001dcc (most recent call first):
  File "C:\Anaconda3\envs\qudi\lib\site-packages\zmq\utils\garbage.py", line 47 in run
  File "C:\Anaconda3\envs\qudi\lib\threading.py", line 916 in _bootstrap_inner
  File "C:\Anaconda3\envs\qudi\lib\threading.py", line 884 in _bootstrap

Thread 0x00001954 (most recent call first):
  File "C:\Anaconda3\envs\qudi\lib\site-packages\zmq\sugar\poll.py", line 99 in poll
  File "C:\Anaconda3\envs\qudi\lib\site-packages\zmq\eventloop\ioloop.py", line 122 in poll
  File "C:\Anaconda3\envs\qudi\lib\site-packages\tornado\ioloop.py", line 862 in start
  File "C:\Anaconda3\envs\qudi\lib\site-packages\zmq\eventloop\ioloop.py", line 177 in start
  File "C:\Anaconda3\envs\qudi\lib\site-packages\ipykernel\iostream.py", line 72 in _thread_main
  File "C:\Anaconda3\envs\qudi\lib\threading.py", line 864 in run
  File "C:\Anaconda3\envs\qudi\lib\threading.py", line 916 in _bootstrap_inner
  File "C:\Anaconda3\envs\qudi\lib\threading.py", line 884 in _bootstrap

Thread 0x00001de4 (most recent call first):
  File "C:\Anaconda3\envs\qudi\lib\threading.py", line 295 in wait
  File "C:\Anaconda3\envs\qudi\lib\threading.py", line 551 in wait
  File "C:\Anaconda3\envs\qudi\lib\site-packages\IPython\core\history.py", line 834 in run
  File "C:\Anaconda3\envs\qudi\lib\site-packages\IPython\core\history.py", line 60 in needs_sqlite
  File "<decorator-gen-23>", line 2 in run
  File "C:\Anaconda3\envs\qudi\lib\threading.py", line 916 in _bootstrap_inner
  File "C:\Anaconda3\envs\qudi\lib\threading.py", line 884 in _bootstrap

Thread 0x00001044 (most recent call first):
  File "C:\Anaconda3\envs\qudi\lib\socket.py", line 205 in accept
  File "C:\Anaconda3\envs\qudi\lib\site-packages\rpyc\utils\server.py", line 128 in accept
  File "C:\Anaconda3\envs\qudi\lib\site-packages\rpyc\utils\server.py", line 241 in start
  File "C:\Software\qudi\core\remote.py", line 214 in run

Current thread 0x000005f0 (most recent call first):
  File "C:\Anaconda3\envs\qudi\lib\site-packages\numpy\core\fromnumeric.py", line 57 in _wrapfunc
  File "C:\Anaconda3\envs\qudi\lib\site-packages\numpy\core\fromnumeric.py", line 1546 in nonzero
  File "C:\Software\qudi\gui\confocal\confocalgui.py", line 830 in get_xy_cb_range
  File "C:\Software\qudi\gui\confocal\confocalgui.py", line 1541 in refresh_xy_image
  File "C:\Software\qudi\gui\confocal\confocalgui.py", line 1524 in update_xy_cb_range
  File "C:\Software\qudi\gui\confocal\confocalgui.py", line 1507 in shortcut_to_xy_cb_centiles
  File "C:\Software\qudi\qtwidgets\pyqtgraphmod\SpinBox.py", line 364 in emitChanged
  File "C:\Software\qudi\qtwidgets\pyqtgraphmod\SpinBox.py", line 370 in delayedChange
  File "C:\Anaconda3\envs\qudi\lib\site-packages\pyqtgraph\SignalProxy.py", line 71 in flush
  File "C:\Anaconda3\envs\qudi\lib\site-packages\pyqtgraph\ThreadsafeTimer.py", line 41 in timerFinished
  File "C:\Software\qudi\core\__main__.py", line 310 in <module>
  File "C:\Anaconda3\envs\qudi\lib\runpy.py", line 85 in _run_code
  File "C:\Anaconda3\envs\qudi\lib\runpy.py", line 193 in _run_module_as_main
Unexpected return value 3221225477. Exiting.