Ulm-IQO/qudi-legacy #66: In predefinded sequence: make an option which lets you chose if blocks are written to gui or not

drogenlied, 2017-02-02:
Migration: trac#71.

Now the GUI editor is always updated with the created sequence. But no choice can be made to NOT load into editor.

Neverhorst, 2017-02-02:
Is that really desirable after all? People are always complaining that they want to scroll through the sequence and check if everything is as expected. This is why the predefined sequences are now automatically loaded into the editor. You can always change to a previously generated block/ensemble/sequence by selecting it and pressing load in the editor side bar.

Neverhorst, 2017-04-25:
I think we can close this issue due to inactivity. No one responded here (also not before on trac) so I think the reporter lost interest in the issue.

closed 2017-04-25 by Neverhorst