Ulm-IQO/qudi-legacy #79: The "current_loaded_asset" instance variable in the pulser hw modules needs to be properly initialized.

drogenlied, 2017-02-02:
Migration: trac#155.

Currently it is by default set to None regardless if the AWG has already something loaded into its channels prior to firing up the software.
So in the activation of the pulser hardware modules the actually currently loaded asset name needs to be extracted.

This was causing an error when having something loaded on the AWG, then closing the software, opening it again and directly starting a measurement. (complained about not having the asset parameters like number of laser pulses).

drogenlied, 2017-02-02:
Is fixed for AWG 70k. Can not test it on other hardware.

Neverhorst, 2018-01-26:
Apparently no issue anymore (or of no interest) for people not using AWG70k so let's close that issue.

closed 2018-01-26 by Neverhorst