Ulm-IQO/qudi-legacy #82: Automatically write sample shift to file

drogenlied, 2017-02-02:
Migration: trac#161.

Write sample shift data to file in real-time, so that it is logged even if qudi crashes or is closed without remembering to save this information.

It is probably important to split the shift data into sensible chunks (ie, start new file when New ROI is clicked).

liuyanulm, 2020-10-07:
This feature has been added in Qudi.

Neverhorst, 2020-10-07:
This has not been implemented, yet.
It's not about saving the sample shift in a `StatusVar` but about saving it occasionally while the module is still active.

See issue #223

liuyanulm, 2020-10-07:
sorry. I misunderstood this. because the poi-drift table is alway updated in my poi-manager gui and I just thought it was already saved in a cache file. I will be more careful when closing issues.

liuyanulm, 2020-10-07:
@Neverhorst is there any qudi module so far can save Status Variables when a crash occurs to Qudi?