Ulm-IQO/qudi-legacy #83: Embed metadata in PDF and PNG figures saved from qudi

drogenlied, 2017-02-02:
It would be good for the figures saved straight from qudi to boast about the awesome software!
Some metadata like "saved from qudi, developed by ..."

kay-jahnke, 2017-02-08:
Implemented the following Metadata for PDF-Files in commit 74fe1f1 :

    d['Title'] = 'Confocal Image'
    d['Author'] = 'qudi - Software Suite'
    d['Subject'] = 'Find more information on: https://github.com/Ulm-IQO/qudi'
    d['Keywords'] = 'Python 3, Qt, experiment control, automation, measurement, software, framework, modular'
    d['Producer'] = 'qudi - Software Suite'
    d['CreationDate'] = timestamp
    d['ModDate'] = timestamp

@drogenlied please specify what else you want saved. Other metadata are also possible, even if they are not specified in the standard.

It might make sense to also save the git-version. For now this is only implemented in the managergui (managergui: def getSoftwareVersion()). If you could move the function somewhere sensible, I could include that in the metadata.

kay-jahnke, 2017-02-08:
Metadata for PNG seems quite tricky:

- metadata is not standardised for PNG and most programs don't recognise it.
- A simple way to add metadata is with the PIL package, which is not maintained anymore and causes more dependencies

**Suggestion: No metadata for PNG.**

drogenlied, 2017-02-09:
The W3C specification explicitly mentions some standard metadata fields: https://www.w3.org/TR/PNG/#11textinfo

However, if it is not easy to convince matplotlib to use these, the suggestion still stands.

a-stark, 2017-02-09:
I also tried to embed a string representation of the measurement array of the data, which works very well, but which makes the pdf file a bit larger then usual (from 15kB to 17 kB for 100 data points), and after all the data could also be inferred from the metadata of the pdf file, which might facilitate eventually the data handling.

I would suggest an additional optional dictionary called 'metadata' which contains string keys and string values. This dictionary is passed as an optional argument to the save logic, which handles the save procedure of the pdf/png metadata.
As a first step, I would suggest to stick to the mentioned standard metadata fields for a png file like @drogenlied mentioned. We need to check whether an implementation without PIL is feasible.

kay-jahnke, 2017-02-11:
Used the active branch Pillow (a fork of PIL) to insert metadata into png files.
The functionality is now working, but I had to extend the dependencies, which I tested and documented. Therefore I create a new branch, since I did not want to destroy any running environment.

Could you please merge the branch save-enhancement back to master, once you list qudi to a new version and have tested it.

Also, in the moment the metadata is static. If we want something dynamic, we should open a new issue and specify, what exactly we want.

a-stark, 2017-02-14:
I think that using [Pillow](https://github.com/python-pillow/Pillow) is a reasonable idea, at least it seems that it is actively developed.

kay-jahnke, 2017-02-14:
OK, lets call this issues resolved for now.

@drogenlied , @a-stark : How do we proceed? Do I just merge back to master, or will you merge back once you publish version 0.8 ton not break other people's dependencies?

drogenlied, 2017-02-14:
People working with master can't really expect much more than a notice on breaking changes.
The more important point is updating the conda environment file, so the environment installation script installs the right thing.

kay-jahnke, 2017-02-14:
This I have done.

So I merge back?

a-stark, 2017-02-14:
If I understand you correctly, you merged at first the master branch in your 'save enhancement' branch. If that was working well, I do not see any problems in committing the changes to the master branch (merge your branch in master).

latchr, 2017-02-14:
If the only problem is a need to add a new python package, then this is
easy to solve. It sounds like the conda env file has already been updated.
When you merge the branch to master I'll send an announcement email to the
institute warning them that it may be necessary to update their qudi conda
environment.

More generally, it would be nice to have a way to make these announcements
directly through Qudi. I'll open a separate issue for the discussion.

a-stark, 2017-02-14:
I would have one remark. Currently the title of the image is set to 
`d['Title'] = 'Confocal Image'`,
which is not true for all pictures. I know that we should open another issue for that (to implement the passing of metadata to savelogic), but until that, we could set a placeholder like
`d['Title'] = 'Image produced by qudi'`.

kay-jahnke, 2017-02-14:
Will do that before I merge.

Give me a few more minutes.

kay-jahnke, 2017-02-14:
solved in commit  95b5741

closed 2017-02-14 by kay-jahnke