Ulm-IQO/qudi-legacy #87: Implement proper handling of the pp-Voltage for the sequence_generator_logic

drogenlied, 2017-02-02:
This is currently an attribute of the logic and needs to be properly set from the GUI.

Neverhorst, 2018-08-08:
Well sort of... The SequenceGeneratorLogic is now able to handle this parameter properly and upon activation the voltages are read from the device.
There is however a settings dialogue missing in the GUI to edit these values per channel.
I postponed and then forgot this point because this raises the question if ALL widgets for "hardware settings" of the pulse generator should be residing in a separate settings dialogue. (adding the pp-voltage widgets is too much for the main window)

Neverhorst, 2018-08-10:
Introduced the above mentioned dialog box. So #384 will fix this issue.

Neverhorst, 2018-09-04:
Fixed by PR #384

closed 2018-09-04 by Neverhorst