Ulm-IQO/qudi-legacy #93: separate ensemble and sequence generation in predefined methods the

drogenlied, 2017-02-02:
Migration: trac#216.

The predefined methods have a generation method and an upload method, which is connected to the generation and upload of only pulse block ensembles! In order to use pulse sequence generation, one needs either a separate file, where the sequence creation methods are present, or the functions, which builds dynamically the creation methods in the GUI, have to distinguish between ensemble creation and sequence creation (to my mind, that can only be done by names, i.e. attaching a suffix like _seq to the generation method.)

drogenlied, 2018-08-08:
I think this is fixed or will be fixed with #384.

Neverhorst, 2018-09-04:
Fixed by PR #384

closed 2018-09-04 by Neverhorst