Ulm-IQO/qudi-legacy #131: How to design the "default estimator" case?

latchr, 2017-02-06:
In the new design, every make_*_fit function will be expecting an estimator to be passed to it.  How do we handle the situation where estimator=None?

latchr, 2017-02-06:
A solution has emerged: for each `make_<fit name>_fit` function we pass units=None and estimator=None.  Then inside this function we have a simple conditional statement where a default estimator is chosen explicitly *by the person who writes the make_fit function*.  For example,

```
if estimator is None:
    error, params = self.estimate_gaussianlinearoffset_peak(x_axis, data, params)
else:
    error, params = estimator(x_axis, data, params)
```

Is this a good solution?

Neverhorst, 2017-02-06:
I think that's a good solution if you want to have the chance for a successful fit even without giving an estimator. (warning should be logged though)
The other not so flexible way would be that the fitting routine will return an error value which can be checked by the calling instance. But I think the first solution is better. (as long as it won't lead to crashes)

closed 2017-02-06 by latchr