Ulm-IQO/qudi-legacy #133: Remove "logging" module import and usage from fit methods modules

Neverhorst, 2017-02-06:
The fit methods modules (linearmethods, lorentzianlikemethods etc.) are still importing and using the old, obsolete logging module from previous qudi versions. This will most probably result in an error when the lines containing things like "logger.warning('Issue this warning!')" are executed.

So in all fit modules the import lines similar to:
`import logging`
`logger = logging.getLogger(__name__)`
have to be removed.

Also all lines similar to
`logger.warning(...)`
have to be changed to
`self.log.warning(...)`

christoph-mueller, 2017-02-08:
changes done

a-stark, 2017-02-08:
I was wondering why the logging module was removed from the fitmethods? In fact, this way of using the logging module should not interfere with the qudi logger, since it will be collected from the core logger of qudi. More information about that can be found [in the documentation](https://ulm-iqo.github.io/qudi-generated-docs/html-docs/logger_usage.html).
If logger.* is replaced by the qudi's internal log.* call, then fit_logic_standalone.py in the tools directory cannot be used and will throw an errors in case of using this attribute. However, this can be solved very quick in attaching the logger as an attribute in the fit_logic_standalone.py init method.

The drawback of this procedure is that you will not be able to locate the exact fitmethod file in the log message, since all messages will be related to the main module FitLogic (which is situated in fit_logic.py). It might compromise the users ability to search for the specific error log.

Neverhorst, 2017-02-08:
Well, we put some effort into changing it in every module back when this logging change has been introduced to qudi. So the idea was why are the fit modules any different than any other module? The argument about better bug-tracking is of course right but since all methods are attached to FitLogic I got the feeling that this is not really a false behavior if the logic is reporting errors (warnings, infos, debugs).

Standalone modules are in my opinion no reason to alter qudi conventions.

If this was a wrong step we should discuss it in the software meeting. I just opened this issue mainly to discuss it.

a-stark, 2017-02-09:
Essentially, I am not against the general concept and I agree with @Neverhorst.
The emphasis was maybe more on the question, whether we want to maintain a fit_logic_standalone.py, which works without direct inheritance of qudi logic base classes (and therefore without the log attribute) and which serves as a test script for the fit methods?
The other part of the comment was more towards the flexible logging behavior of qudi.

jochen-scheuer, 2017-02-09:
In my opinion the fit logic standalone is very useful! Like this the postprocessing of data can be done with the same fits and while postprocessing changes and improvements of the fit logic will also contribute to qudi!

Neverhorst, 2017-02-09:
Agree, @jochen-scheuer, and it's not about abandoning the standalone module. I suggest we stick to the convention with self.log.<messagetype>() like in any other module and just attach the logger in the __init__() of the standalone module. If we make an exception by explicitly importing the logger, we should do it in the standalone modules only. (because they are not really a part of qudi)

Neverhorst, 2017-02-13:
Just commited the proposed changes to the standalone module. Seems to work. (see commit 683f31c673aa5d79714a642c4990bf1d6cdad502)

a-stark, 2017-02-14:
I would also agree with this procedure. I think we could prevent long searchings for a info/warning/error message in the code, if the messages are kept not too general (like 'fit method failed'), but rather more specific for each fit method (like: 'the two lorentz fit failed'). Typo errors or other exceptions, which might happen within a method contain anyway a proper and explicit Traceback to the file and linenumber.

I would close this issue today, if nobody has another opinion.

commit 2a6ff40a (2016-05-10): Change N14 and N15 estimator such that it works now with SI and also with not sensible values (If the hyperfine is not resolvable).

Estimating the N14 and N15 in a new way such that the FWHM/sigma and amplitude are calculated via an integral of the data.

fixed #133
closed 2017-02-14 by a-stark