Ulm-IQO/qudi-legacy #135: Properly handle the methods in fitmethods/*.py that we do not need to import to fit_list

latchr, 2017-02-07:
There are various methods like gaussian and xlogy and so on that are defined in the fitmethods/*.py files but are not needed in fit_list.  Currently the list is generating properly (excluding these other methods) but they give errors like the following (also with dialog widget):

```
Loading Qudi...
============= Starting Manager configuration from /home/lachlan/physics/code/qudi/config/example/default.cfg =================

============= Manager configuration complete =================

Remote connection not secured! Use a certificate!
sIt was not possible to import element Model into FitLogic.
It was not possible to import element Model into FitLogic.
It was not possible to import element Model into FitLogic.
It was not possible to import element OrderedDict into FitLogic.
It was not possible to import element Parameters into FitLogic.
It was not possible to import element gaussian into FitLogic.
It was not possible to import element InterpolatedUnivariateSpline into FitLogic.
It was not possible to import element Model into FitLogic.
It was not possible to import element OrderedDict into FitLogic.
It was not possible to import element Parameters into FitLogic.
It was not possible to import element Model into FitLogic.
It was not possible to import element InterpolatedUnivariateSpline into FitLogic.
It was not possible to import element Model into FitLogic.
It was not possible to import element Parameters into FitLogic.
It was not possible to import element gammaln into FitLogic.
It was not possible to import element gaussian into FitLogic.
It was not possible to import element poisson into FitLogic.
It was not possible to import element xlogy into FitLogic.
It was not possible to import element ConstantModel into FitLogic.
It was not possible to import element GaussianModel into FitLogic.
It was not possible to import element InterpolatedUnivariateSpline into FitLogic.
It was not possible to import element Model into FitLogic.
It was not possible to import element OrderedDict into FitLogic.
It was not possible to import element Parameters into FitLogic.
```

This is happening from an `try...except` in the code in fit_logic.py.  How can we make this smarter so that the user is not faced with "errors"?

Neverhorst, 2017-02-07:
But I think the dictionary does not look as intended. It's built the same way as before but is it right that you have as "fit_methods" something like "doublelorentzoffset" AND "doublelorentzoffset_N15"? I thought the latter is just a custom estimator/parameter set but the same fit.

jochen-scheuer, 2017-02-07:
you are right, it should look like doublelorentzoffset

Neverhorst, 2017-02-07:
Just a moment...

Neverhorst, 2017-02-07:
With commit e7094de2aa3fcfbd32a5be1a5a585d30d5385a87 the import of the fit modules is now much smarter but also more unforgiving. Meaning it expects for all "make_<fit_name>_fit" methods also a corresponding "make_<fit_name>_model" and "estimate_<fit_name>" method. This is necessary if all methods in the fit_list dictionary should have "make_fit", "make_model" and "generic" (estimator) as keys. I think this is the minimal set of methods for each fit. Additional estimators (like N14 etc.) can be of course added.
Now it throws a lot of errors but is asking specifically for the missing methods.
In order to avoid code duplication one could make "dummy methods" that only call another method.

I see no other way to enforce a common minimal set of methods for each fit in fit_list while being able to automatically create this fit_list from method names. But if anybody has any ideas...

latchr, 2017-02-08:
This solution seems best for now.

closed 2017-02-08 by latchr