Ulm-IQO/qudi-legacy #136: POI Manager throws fatal error when setting a new POI

stefanhaeussler, 2017-02-08:
Error message:

AttributeError: 'PoI' object has no attribute 'log'
Traceback (most recent call last):

  File "C:\Software\qudi\gui\poimanager\poimangui.py", line 596, in set_new_poi
    key = self._poi_manager_logic.add_poi()

  File "C:\Software\qudi\logic\poi_manager_logic.py", line 267, in add_poi
    new_track_point = PoI(point=position, key=key)

  File "C:\Software\qudi\logic\poi_manager_logic.py", line 64, in __init__
    self.log.error('Given position does not contain 3 '

latchr, 2017-02-08:
I have tested this with the default config (dummy hardware) and the current revision of master ( be26d0d ). I cannot reproduce this bug.  

Can you check whether this problem still occurs after a git pull to the current revision?

Neverhorst, 2017-02-08:
Same here. Can't reproduce with my experimental setup nor dummy.

stefanhaeussler, 2017-02-09:
Works properly if you deactivate the fourth analog output channel, so only with 3 active channels.

latchr, 2017-02-12:
What is the 4th channel used for?  Is it important for its position to be saved as part of the POI, or is it only important to have the 3 "positional" values?

stefanhaeussler, 2017-02-13:
In our case it is used to scan the length of our cavity. Other experiments use it for example to scan their lasers. I think it is not needed (at least in our case) to save its position in the POI.

latchr, 2017-03-22:
I have re-investigated this and discovered a subtle problem we need to fix.  Somehow in the change over to the new logging system we have not correctly given the PoI class the log attribute. The original traceback highlights this.  I'm sure this is easy to fix, but I cannot see how the other classes are getting log.  @tobiasgehring, @a-stark, @drogenlied can you help me out?

a-stark, 2017-03-22:
Hi,  the Qudi logger comes from the `GenericLogic` and eventually from the `Base` class. There are two solutions how to handle this.
One is to inherit `GenericLogic` into the `class PoI`. 
If that is not desired, then you can use a child logger by
```
import logging
logger = logging.getLogger(__name__)
```
and insert in the ``__init__`` method:
```
self.log = logger
```
or if you do not want to store it as an attribute of the class just replace `self.log` by `logger`. This should do the job. You can check which scenario fits the best for you.

latchr, 2017-03-23:
I have added a child logger to the PoI class.  I also added a sanity check in the add_poi method of poi manager logic, which should catch this problem even earlier and clearly tell the user what went wrong.

commit c289c97c (2017-03-23): Added logging to the PoI class in POI manager logic.  Added sanity check to see that 3-D positions are supplied when adding a POI.  Fixed #136
closed 2017-03-23 by latchr