Ulm-IQO/qudi-legacy #138: Add fit parameters display to optimizer widget in confocal gui

latchr, 2017-02-08:
It would be good for the gui to display the fit parameters for the optimizer (especially width, the others are probably not so important).  For example, this would help with the alignment process of a confocal microscope where the user wishes to make the sample surface as "thin" as possible in the depth scan.

latchr, 2017-02-08:
This could be in its own displaywidget so that it can be folded away when not needed.

LukePiewalker, 2020-10-29:
Is this issue outdated now @Neverhorst @latchr? In the current version of the confocal gui the fit values are displayed as: (x,y,z):  &#956; = ( ... , ... , ... ) &#956;m  &#963; = ( ... , ... , ... ) &#956;m  ![\Large x=\frac{-b\pm\sqrt{b^2-4ac}}{2a}]
In the scanner_gui of the new core there had been placeholders, however after pulling the current status today, they are gone?

Neverhorst, 2020-10-29:
You are right @LukePiewalker, this issue had been resolved a long time ago by the mentioned fit parameter display.

The new core is still WIP and as such it can change quite drastically. The parameter display will be included in the release version one way or the other.

closed 2020-10-29 by Neverhorst