Ulm-IQO/qudi-legacy #142: Find a sensible way to categorise and order the fit methods in fitlogic.fit_list

latchr, 2017-02-08:
These are added to a dictionary probably in the order they are encountered while reading a file.  Dictionaries are inherently un-ordered.  It would really help the user find what they need if the fit_list dictionary could be read with some sensible order.

Neverhorst, 2017-02-08:
If it is just about having an alphabetic order in the combobox... just do something like `methods = list(fit_list)` and `methods.sort()` and then add this "methods" list to the combobox. For accessing the dictionary once you select a key from the combobox the complexity is always 1 so no need to sort the actual dictionary. One could of course also sort the dict upon creation... I will do that now... costs (almost) nothing. ;-)

Neverhorst, 2017-02-08:
Done with commit 514d73c902972c096ca9df08fb1dd3cedf3a4e08

closed 2017-02-08 by Neverhorst