Ulm-IQO/qudi-legacy #143: cannot start pulsed measurement after running an ensemble with one laser pulse

cyz75, 2017-02-08:
I tryed a simple sequence with one bright and one dark pulse, with alternating. An error related with FFT occured. Now I cannot start any pulsed measurement without this error:

`ValueError: Invalid number of FFT data points (0) specified.
Traceback (most recent call last):

  File "C:\HYPERDIAMOND\Software\qudi\logic\pulsed_measurement_logic.py", line 793, in _pulsed_analysis_loop
    self._compute_fft()

  File "C:\HYPERDIAMOND\Software\qudi\logic\pulsed_measurement_logic.py", line 1127, in _compute_fft
    self.signal_fft_y2 = np.abs(np.fft.fft(corrected_y2))[:middle]

  File "C:\Anaconda3\envs\qudi\lib\site-packages\numpy\fft\fftpack.py", line 189, in fft
    output = _raw_fft(a, n, axis, fftpack.cffti, fftpack.cfftf, _fft_cache)

  File "C:\Anaconda3\envs\qudi\lib\site-packages\numpy\fft\fftpack.py", line 55, in _raw_fft
    % n)
`

Even restarting doesn´t help, but when loading pulsedmeasurementgui I get the following error:

IndexError: index 0 is out of bounds for axis 0 with size 0
Traceback (most recent call last):

`File "C:\HYPERDIAMOND\Software\qudi\gui\pulsed\pulsed_maingui.py", line 1942, in measurement_sequence_settings_updated
    self._pa.ana_param_x_axis_start_ScienDSpinBox.setValue(controlled_vals[0])`  

Deleting all pulsed log files fixed it

Neverhorst, 2017-02-08:
Tracked the issue to a false user input. This problem occurs if you enter "1" for number of lasers in the pulse analysis tab but also tick "alternating?". An alternating sequence always has to have an even number of laser pulses. (otherwise it wouldn't be alternating right?)

Nevertheless this use case should be caught by sanity checks and handled properly without breaking anything. Will work on that.

Neverhorst, 2017-02-08:
Should be fixed with commit 74fe1f15c633f6a7d31a368b3c45e06f6413fceb.
Please check and close issue if it's working.

closed 2017-02-15 by Neverhorst