Ulm-IQO/qudi-legacy #146: saving in pulsedmeasurement: additional info into header

cyz75, 2017-02-10:
I´m not sure if this already is a issue or if there is a way to configure it by my self, but some information would be good to be added to the saved data.
- Measurement time (in header)
- errors ( additional row)
- result of the current fit (in header)

Some more points that are interesting but not necessary:
Sequence parameters:
-Block Name
-Sampling Rate
-AWG_model
Also I don´t know what info about the sequence is saved but as an example for Rabi useful information could look like
#Block Name: Rabi
#Frequency(GHz): 2.87 GHz
#Amp: 0.5 V
#Start value (ns): 4.0
#Increment (ns): 4.0

Neverhorst, 2017-02-13:
Commit e72aa3026ee3ac0345cd8c99d68c6c56c64e24bc introduced the saving of the error bars in the plot and in the data file (additional rows - 1 for non-alternating, 2 for alternating). Pressing save in the GUI will only save errors if the "show errorbars" checkbox is checked. For scripts the save_measurement_data method in pulsed_measurement_logic.py now has an additional optional parameter "with_error=True" that is used to determine whether to save errors or not.

Neverhorst, 2017-02-14:
Now also the approx. measurement time (in the logic the attribute "elapsed_time") is saved in the parameters header. See commit 3837cb158b73bd3435c5e15c2090389427b7dbc4.