Ulm-IQO/qudi-legacy #148: The "double exponential decay" methods in fitlogic are gaussian, not double exponential

latchr, 2017-02-12:
A double exponential function is of the form y = a^(b^x) .  It is literally "doubly-exponential".  See more at [the Wikipedia article](https://en.wikipedia.org/wiki/Double_exponential_function).

Currently we have fit methods which are called "double exponential" but have the form y=e^-(x^2) .  This is the stretched exponential with stretching exponent of 2 (which is exactly how the fit model is created).  However, this special case is a Gaussian function.

Where is this wrongly-named "double exponential"/Gaussian used?

Neverhorst, 2017-02-13:
I'm not quite sure what the purpose of this issue is. Should we go and rename all methods?

latchr, 2017-02-13:
Firstly, I'm wanting to find out who uses the "double exponential decay"
fit and what the context is.  I don't think we should rename the methods -
instead I recommend we delete the "double exponential decay" fit methods
because they are simply a half-gaussian peak.  However, if the context is
really common and we'd rather have a "lifetime" parameter than
half-width-half-maximum, then we can rename it to be "gaussian decay" or
something like that.

kay-jahnke, 2017-02-13:
Hi guys,

for your understanding: I think the name might stem from the very early time of the institute. At the beginning of my diplom thesis (2011) we worked on the T1/T2 decay times of NV spins in polycrystalline diamond. These early papers used the expression double exponential decay, because a gaussian would suggest a mirror symetry that isn't there.

I think by now this expression is not actively used anymore. But best ask Boris, Fedor and a few of the spin decay guys.

It might actually have been me, who introduced this fit and I think it can be renamed. However I have no stakes in the matter and you guys have to work with it in the future.

Cheers,
Kay

a-stark, 2017-02-14:
I think that it was me who introduced this naming and fit in qudi, but it is not used in qudi so far (just in my personal scripts). The main difference to the gaussian fit is the estimator and a slightly different model.

<a href="https://www.codecogs.com/eqnedit.php?latex=f_{\text{decay}}(x)&space;=&space;\text{offset}&space;&plus;&space;\text{amplitude}&space;\cdot&space;e^{-&space;(\frac{x}{\text{lifetime}})^{2}&space;}\\[0.5cm]&space;g_{\text{gauss}}(x)&space;=&space;\text{offset}&space;&plus;&space;\text{amplitude}&space;\cdot&space;e^{-&space;\frac{1}{2}(\frac{&space;x&space;-&space;\text{center}}{\sigma})^2&space;)}" target="_blank"><img src="https://latex.codecogs.com/gif.latex?f_{\text{decay}}(x)&space;=&space;\text{offset}&space;&plus;&space;\text{amplitude}&space;\cdot&space;e^{-&space;(\frac{x}{\text{lifetime}})^{2}&space;}\\[0.5cm]&space;g_{\text{gauss}}(x)&space;=&space;\text{offset}&space;&plus;&space;\text{amplitude}&space;\cdot&space;e^{-&space;\frac{1}{2}(\frac{&space;x&space;-&space;\text{center}}{\sigma})^2&space;)}" title="f_{\text{decay}}(x) = \text{offset} + \text{amplitude} \cdot e^{- (\frac{x}{\text{lifetime}})^{2} }\\[0.5cm] g_{\text{gauss}}(x) = \text{offset} + \text{amplitude} \cdot e^{- \frac{1}{2}(\frac{ x - \text{center}}{\sigma})^2 )}" /></a>

One could introduce a new parameter lifetime = sqrt(2)* sigma, in the gaussian model to cope for that and set center=0 as a constraint. But judging from the model for 'double exponential', and how it is created, I would rather think that it belongs to the decay methods (which describes a certain decay process). 
I have a feeling that it becomes more complicated, if it is included in the gaussian methods. 

I agree that the name 'double exponential' is misleading, does anyone has a better idea?

latchr, 2017-03-02:
I think the best way to avoid the bad naming is simply to call it what it is: a stretched-exponential with the stretch-factor beta=2.  This means that we do not need to give it a special make_*_fit method.  Instead the user can select stretched-exponential fit and simply set beta=2 with vary=false.

Is there a way for the estimator to set this beta=2 and vary=false?  That would be the most elegant way to do this, with the fit method being "stretched exponential" and then selecting an estimator called "beta=2" or "gaussian" or something.

latchr, 2017-03-15:
This has been resolved by removing the special case for beta=2.  Now if you want the fit that used to be called "double exponential decay" you must use "stretched exponential" and force the stretch factor beta=2.

commit 814248be (2016-05-15): Use default parameter either from the logic or if gui related, then save in GUI. That fixed #148.
closed 2017-03-15 by latchr