Ulm-IQO/qudi-legacy #152: Does set_up_line() needs to be in the ConfocalScannerInterface?

Neverhorst, 2017-02-20:
It appears to me as if the interface method set_up_line() is just a helper method for the method scan_line(). Only the latter needs to be accessed by external logic modules yet both are part of the ConfocalScannerInterface. Is that right?

drogenlied, 2017-02-20:
That seems to be the case.
set_up_line does not need to be in the interface or any interfuse and is purely internal.

kay-jahnke, 2017-02-20:
Indeed, this function is only called by the ni-hardware itself (and the same for the dummy). So it is not needed in the interface, especially since its function is quite trivial.

So I deleted the function from the interface and made it private in ni_card and dummy_scanner by adding a leading underscore. Tested dummy, and everything works fine in commit b37a7d9.

Issue closed!

closed 2017-02-20 by kay-jahnke