Ulm-IQO/qudi-legacy #153: Make module state machine more sane

drogenlied, 2017-02-22:
Right now, `locked` modules do not get deactivated when quitting Qudi.

According to the documented model, quitting Qudi should not be possible while modules are `locked`, right now it is just terminated regardless.

However, this behavior has proven undesirable in day-to-day use.

The proposed new model would allow deactivation from `idle`, `running` and `locked` states with a pop-up dialog/console input prompt for confirmation when `locked` modules are present.

At the same time, we can get rid of the `blocked` state, which is not in use.

drogenlied, 2017-02-22:
Check if 0317fc95ea631a27ae3f57b37377ce6f3de9e760 is satisfactory.

Neverhorst, 2017-02-23:
Doesn't seem so... It is correctly prompting when a module is locked and you want to quit but no deactivation is reached if you click on "OK".

drogenlied, 2017-02-23:
Ok now it does and there is a nice error when deactivating counter logic from the locked state.

Neverhorst, 2017-02-23:
Hmm, can't confirm that. Everything seems just the same. After confirming forced quit no deactivation is called and nothing happens. Tried on dummy and real setup (Ubuntu/Windows).

Neverhorst, 2017-02-23:
Could it be that the signal "sigShutdownAcknowledge" from manager is not connected? This is fired when you are using the GUI. Otherwise the realQuit() will be called directly.

drogenlied, 2017-02-23:
The signal is definitely connected, otherwise you would not be seeing the confirmation popup.

When I uncomment the print in the deactivation in counterlogic, I can see the logic going locked -> deactivated for a running counter after the prompt and idle->deactivated for a stopped counter.

Neverhorst, 2017-02-23:
Ok, we narrowed it down to an issue with QtWidgets.QMessageBox.question. Apparently the default in Qt5 is a yes/no-dialogue while in Qt4 it's an OK-dialogue. Fixed it by explicitly changing that to Yes/No (like the dialogue box for asking to reload qudi).

Encountered that the hardware modules are actually deactivated before logic and GUI which is a bug but this will be discussed in another issue.

closed 2017-02-23 by Neverhorst