Ulm-IQO/qudi-legacy #173: Public mailing list

drogenlied, 2017-03-20:
Find a hosting service for a public Qudi development and announcement mailing list, as the Ulm University list system will not support external members.

latchr, 2017-03-21:
Is there a way to have a one-off dialog box that opens after a major revision change, to explain changes to the code or config?  Ideally this would be something that could be flagged to appear after a major pull request is merged for example?

Perhaps there could be a metadata file created and stored that allows Qudi to know what the "version" was the last time it was run.  That would allow it to know if major changes had taken place.

latchr, 2017-03-23:
This looks quite suitable: http://www.freelists.org/

Any objections?

drogenlied, 2017-03-25:
It is seems to work for the DokuWiki and Haiku projects so it should be good enought for us.

drogenlied, 2017-03-27:
I have applied for 2 lists, qudi-dev for discussion and qudi-announce for version and master change announcements.

drogenlied, 2017-03-28:
For development-related questions and discussion, please use the [qudi-dev mailing list](http://www.freelists.org/list/qudi-dev).

If you just want updates about releases and breaking changes to Qudi without discussion or issue reports, subscribe to the [qudi-announce mailing list](http://www.freelists.org/list/qudi-announce).

a-stark, 2017-03-28:
Nice job Jan, I have checked also other mailing lists, but it seems that freelist is really the best choice. For me the subscription works, so let's move to the qudi-dev mailing list.

closed 2017-03-28 by drogenlied