Ulm-IQO/qudi-legacy #195: Doublepoissonian fit still not propery implemented

Neverhorst, 2017-04-27:
While checking out how many fits have human readable return params I encountered a problem with poissoniandouble fit. Apparently the fit methods modules are missing a method (or wrong naming).

Error message:
Traceback (most recent call last):
  File "C:\Development\Software\qudi\gui\fitsettings.py", line 114, in buttonClicked
    self.applySettings()
  File "C:\Development\Software\qudi\gui\fitsettings.py", line 245, in applySettings
    model, params = self.all_functions[widget.fit]['make_model']()
  File "C:\Development\Software\qudi\logic\fitmethods\poissonianlikemethods.py", line 148, in make_poissoniandouble_model
    return self.make_multiplepoissonian_model(2)
  File "C:\Development\Software\qudi\core\base.py", line 138, in __getattr__
    return Fysom.__getattr__(self, name)
  File "C:\Development\Software\qudi\core\FysomAdapter.py", line 36, in __getattr__
    self.__class__, name))
AttributeError: '<class 'logic.fit_logic.FitLogic'>' object has no attribute 'make_multiplepoissonian_model'

a-stark, 2017-04-27:
I am currently working on related parts of that issue and as a next step to even simplify the creation of such a human readable parameters. The branch for that is the [utilize_user_data_in_fit](https://github.com/Ulm-IQO/qudi/tree/utilize_user_data_in_fit), which will require the most recent version of [lmfit](https://github.com/lmfit/lmfit-py/_ (0.9.6) containing my pull request to lmfit.
The basic idea here, new parameters should be specified already in the model, rather then in the fit method. I am going to be away for 2 weeks but after that I will make the pull request ready.

drogenlied, 2017-05-12:
fixed in #201

Neverhorst, 2017-05-17:
Will close that one as soon as #201 has been merged into master. The rest (human readable return value) is then a duplicate of #132.

closed 2017-05-19 by Neverhorst