Ulm-IQO/qudi-legacy #196: Loading ROI does not work anymore

cyz75, 2017-04-28:
I wanted to load a ROI
What I´ve done step by step:
1. Press Open ROI
2. selcted C:\Data\2017\01\20170127\ROIs\20170127-1343-20_Sample69_NV21.dat and open
3. Error: 
FileNotFoundError: [Errno 2] No such file or directory: 'C'
Traceback (most recent call last):

  File "C:\HYPERDIAMOND\Software\qudi\gui\poimanager\poimangui.py", line 921, in load_roi
    self._poi_manager_logic.load_roi_from_file(filename=this_file)

  File "C:\HYPERDIAMOND\Software\qudi\logic\poi_manager_logic.py", line 735, in load_roi_from_file
    roifile = open(filename, 'r')

Neverhorst, 2017-04-28:
When I save ROIs the filename formatting is slightly different. The currently selected POI name (in your case NV21) comes right after the date. Has the file format of the saved ROIs changed since beginning of this year and is causing that error, @latchr? I can't reproduce that.

Neverhorst, 2017-04-28:
Bug was not actually a bug but an old qudi conda environment (using PyQt4, see also issue #197). Updating to the latest environment fixed it.

closed 2017-04-28 by Neverhorst