Ulm-IQO/qudi-legacy #197: using zoom in confocal gives error

cyz75, 2017-04-28:
Using the zoom tool is producing a error. The image changes anyway but the scan range isn´t changed

AttributeError: 'QMouseEvent' object has no attribute 'localPos'
Traceback (most recent call last):

  File "C:\HYPERDIAMOND\Software\qudi\gui\confocal\confocalgui.py", line 1952, in xy_scan_end_zoom_point
    pos = viewbox.mapSceneToView(event.localPos())

Neverhorst, 2017-04-28:
I don't get errors when zooming but I can confirm that the zoom is definetely not working as intended. If you zoom in by dragging a rectangle in your confocal scan and then start a new scan... 9/10 times strange behaviour related to the scan ranges.

Neverhorst, 2017-04-28:
Ok Error was caused by old qudi environment (using PyQt4) but the ranges are still sometimes confused.

closed 2017-05-03 by cyz75