Ulm-IQO/qudi-legacy #199: Optimizer is not scanning

cyz75, 2017-05-03:
Optimizer isn´t scanning at all
Setup N27 
config:     optimizer:
        module.Class: 'optimizer_logic.OptimizerLogic'
        connect:
            confocalscanner1: 'mynicard'
            fitlogic: 'fit'

cyz75, 2017-05-03:
Changing scan range to 1m solved the problem, which means just the units are messed up. Instead of m it should be µm, also in the GUI

drogenlied, 2017-05-03:
Are you sure that the scan ranges in the config are aof the right size?
They should be something like  200e-6, not 200.

Do not forget to remove the confocal related files from the app_status folder after making these changes or you will still have the image history with strange dimensions.

Neverhorst, 2017-05-03:
I've seen this issue over and over again. It's exactly the two points @drogenlied mentioned.

closed 2017-05-04 by cyz75