Ulm-IQO/qudi-legacy #205: Scanner not working on 2-counter NIDAQ usb-6259

latchr, 2017-05-12:
I'm trying to get Qudi to run on the NI USB-6259, which only has 2 counters.  I have the nicard config:

    mynicard:
        module.Class: 'ni_card.NICard'
        clock_channel: '/Dev1/Ctr0'
        #counter_channel: '/Dev1/Ctr1'
        scanner_clock_channel: '/Dev1/Ctr0'
        scanner_counter_channel: '/Dev1/Ctr1'
        photon_source: '/Dev1/PFI0'
        gate_in_channel: '/Dev1/PFI9'
        counting_edge_rising: True
        odmr_trigger_channel: '/Dev1/PFI9'
        scanner_x_ao: '/Dev1/AO0'
        scanner_y_ao: '/Dev1/AO1'
        scanner_z_ao: '/Dev1/AO2'
        clock_frequency: 100
        scanner_clock_frequency: 100
        x_range:
        - 0e-6
        - 50e-6
        y_range:
        - 0e-6
        - 50e-6
        z_range:
        - -15e-6
        - 15e-6
        a_range:
        - -10e-6
        - 10e-6
        voltage_range:
        - -10
        - 10
        samples_number: 10

Counting works with the slow counter.  Moving the scanner position works (in other words, it seems I can drive the inputs/outputs correctly on this NIDAQ).  However, with the counter not running (actually, with the counter modules not even loaded) I cannot perform a confocal scan.  I get the following 3 errors:

Error while scanning line.
PyDAQmx.DAQmxFunctions.DAQError: The specified resource is reserved. The operation could not be completed as specified.
Task Name: ScannerClock

Status Code: -50103
 in function DAQmxStartTask
Traceback (most recent call last):

  File "C:\LabSoftware\qudi\hardware\ni_card.py", line 1452, in scan_line
    daq.DAQmxStartTask(self._scanner_clock_daq_task)

  File "<string>", line 2, in function

  File "C:\ProgramData\Anaconda3\envs\qudi\lib\site-packages\PyDAQmx\DAQmxFunctions.py", line 29, in mafunction
    raise DAQError(error,errBuff.value.decode("utf-8"), f.__name__)

Error stopping analog output.
PyDAQmx.DAQmxFunctions.DAQError: Finite acquisition or generation has been stopped before the requested number of samples were acquired or generated.
 in function DAQmxStopTask
Traceback (most recent call last):

  File "C:\LabSoftware\qudi\hardware\ni_card.py", line 1037, in _stop_analog_output
    daq.DAQmxStopTask(self._scanner_ao_task)

  File "<string>", line 2, in function

  File "C:\ProgramData\Anaconda3\envs\qudi\lib\site-packages\PyDAQmx\DAQmxFunctions.py", line 34, in mafunction
    raise DAQError(error,errBuff.value.decode("utf-8"), f.__name__)

Could not close scanner counter.
PyDAQmx.DAQmxFunctions.DAQError: Finite acquisition or generation has been stopped before the requested number of samples were acquired or generated.
 in function DAQmxStopTask
Traceback (most recent call last):

  File "C:\LabSoftware\qudi\hardware\ni_card.py", line 797, in close_counter
    daq.DAQmxStopTask(task)

  File "<string>", line 2, in function

  File "C:\ProgramData\Anaconda3\envs\qudi\lib\site-packages\PyDAQmx\DAQmxFunctions.py", line 34, in mafunction
    raise DAQError(error,errBuff.value.decode("utf-8"), f.__name__)

latchr, 2017-05-12:
The first question is whether it is correct to configure the clock and counter channels like this.
    clock_channel: '/Dev1/Ctr0'
    counter_channel: '/Dev1/Ctr1'
    scanner_clock_channel: '/Dev1/Ctr0'
    scanner_counter_channel: '/Dev1/Ctr1'

Is this double-up the right way to handle a NIDAQ with only 2 counters?  Also, do we have this situation described in the documentation anywhere?

Neverhorst, 2017-05-12:
Using the same counters for scanner and slow_counter (so only Ctr0 and Ctr1 like described in your case) everything works on my setup with NI PCIe-6323. Timetrace and Scanning won't work at the same time of course but if you stop the counter, you can scan confocal without any problems.

Neverhorst, 2017-06-20:
Can we close this issue due to inactivity?
The task of improving the counting scheme using NI Cards should be a separate issue.

latchr, 2017-06-26:
I agree.

closed 2017-06-26 by latchr