Ulm-IQO/qudi-legacy #206: Selecting CW in ODMR GUI is not changing MW-hardware into cw-mode

Neverhorst, 2017-05-12:
Instead the microwave device just switches on in list mode causing it an cw-output of the start frequency (since no triggers are generated).
This issue was encountered with Anritsu MW source (max. 20 GHz).
At first glance this is an issue with inconsistent method naming and resulting confusion in GUI, logic and hardware. Not sure what the concept with the MW hardware interface is when you have set_frequency, set_power, set_cw, set_list, set_sweep, sweep_on, list_on and on.
Seems a bit... "not thought through" when it comes to what output mode is active in the device.

drogenlied, 2017-05-12:
No shit, Sherlock!

Neverhorst, 2017-05-17:
Complete microwave hardware/interface and odmr logic/GUI overhaul is work in progress in branch microwave_interface_overhaul since it makes no sense to just fix that specific issue in a quick-and-dirty way and leave the rest as it is.

Neverhorst, 2017-06-12:
Fixed with merge commit 5d021a8f390c26752e0d574696eb106237ddc213

closed 2017-06-12 by Neverhorst