Ulm-IQO/qudi-legacy #225: Possible clash of status variables for different configurations

tobiasgehring, 2017-07-12:
## *What* is affected by this bug?
Status variables are saved in a directory called "app_status" in the same directory as the configuration file. If multiple configuration files are in the same folder, they share the same app_status directory which means that if they use the same modules, status variables are shared between the configurations.

## Expected behavior (i.e. solution)
Status variables for different configurations should be independent of each other.

Possible solution: Rename folder "app_status" into "configuration_filename_status".

a-stark, 2017-07-12:
@tobiasgehring thanks for pointing that out.
If I actually think about it, then I must say that I use this "feature" quite often.
For instance, if I use a different microwave source for measurement A than for measurement B I quickly load the other configuration. The advantage is that other modules, which are the same in both configurations retain their configuration data (e.g. a recorded picture, or a specific set of parameter settings).
Wouldn't it solve the problem if each configuration file is stored in a different folder?
Like so:
```
config-folder
    /user_a
        user_a_config.cfg
        /app_data/
    /user_b
        user_b_config.cfg
        /app_data/
```

Neverhorst, 2018-10-25:
Since this topic is apparently a matter of opinion/taste and nothing has been discussed since over a year I will close this issue.

closed 2018-10-25 by Neverhorst