Ulm-IQO/qudi-legacy #226: Instrument libraries

tobiasgehring, 2017-07-14:
There are a lot of instrument libraries out there trying to interface various scientific instruments. In this issue these shall be reviewed for possible inclusion into qudi with the benefit of enabling access to a larger pool of hardware.

I will start with a list of libraries I found (only ones that appear actively maintained)
- [Instrumental](https://github.com/mabuchilab/Instrumental)
- [InstrumentKit](https://github.com/Galvant/InstrumentKit)
- [pymeasure](https://github.com/ralph-group/pymeasure)
- [Qcodes](http://qcodes.github.io/Qcodes/) -- more a competitor, but also has some instrument drivers
- [qtlab](https://github.com/govenius/qtlab) -- last commit Nov, 9th, 2016

Please add if you know more.

tobiasgehring, 2017-07-15:
# InstrumentKit

- hardware interfaces classes, however the existing ones seem to be very rudimentary and not well designed (yet?)
- very nice abstraction of the actual connection to the hardware + possibility of specifying an URI (nice for specifying the connection in our configuration file)

- still supports python 2.7 (-> requires us to support it if we contribute drivers)
- in some drivers they declare support classes inside a class (I'm not so a fan of that and it confused me at first - I prefer them outside with a leading underscore), in other drivers however they declare them outside (as I prefer)
- includes a Thorlabs APT driver connecting via Thorlabs' communication protocol. However, only a patched version is compatible with python3 and it's far from working with my stage.
- they consider an effort to include instruments implemented in the other libraries mentioned here
- all quantities have associated units (using the quantities package)

- I actually have none of the instruments they have drivers for, so testing is a bit hard

### Dependencies
- numpy
- pyserial
- quantities
- future>=0.15
- enum34
- python-vxi11>=0.8
- pyusb
- python-usbtmc
- pyyaml

quantities is currently incompatible with the newest version of numpy.

drogenlied, 2017-07-17:
License compatibility:
 * Instrumental: GPLv3, no code use or sharing problems
 * InstrumentKit: AGPLv3, do not copy code from here under any circumstances, library use is fine, contributions must be fully your own
 * pymeasure: MIT, code is copyable, needs attribution, contributions must be fully your own
 * Qcodes:  MIT, code is copyable, needs attribution, contributions must be fully your own
 * qtlab: unclear situation, some files have GPLv2+ headers and could be incorporated, contributions must be fully your own

tobiasgehring, 2017-11-18:
[python-ivi](https://github.com/python-ivi/python-ivi): MIT code is copyable, needs attribution, contributions must be fully your own

drogenlied, 2017-12-24:
From a quick glance, python-ivi looks like one could write a general wrapper for it and use the implemented instruments.

tobiasgehring, 2017-12-24:
I'm working on that right now starting with the oscilloscopes,
scope_interface branch. It's not worth checking out right now though.