Ulm-IQO/qudi-legacy #238: Latest pyqtgraph causes issues with custom SpinBox

drogenlied, 2017-08-01:
## *What* is affected by this bug?

When using pyqtgraph from git or pyqtgraph-qo from pypi, any GUI module using ScienDSpinBox or ScienSpinBox with bounds produces exceptions.

## *When* does this occur?

Change the number in any ScienDSpinBox os ScienSpinBox, like the scan range fields in the Confocal GUI.

## *Where* on the platform does it happen?

Confocal example:
```
Traceback (most recent call last):
  File "/home/jan/qo/qudi/qtwidgets/pyqtgraphmod/SpinBox.py", line 575, in editingFinishedEvent
    self.setValue(val, delaySignal=False)  ## allow text update so that values are reformatted pretty-like
  File "/home/jan/qo/qudi/qtwidgets/pyqtgraphmod/SpinBox.py", line 334, in setValue
    if bounds[0] is not None and value < bounds[0]:
TypeError: unorderable types: str() < decimal.Decimal()
```

## *How* do we replicate the issue?

- Install pyqtgraph-qo from pypi.
- Start Qudi with the example configuration
- Open the Confocal GUI module
- Change one of the Scan Range values
- hit Tab or Enter to trigger editingFinished()

## Expected behavior (i.e. solution)

No exception.

## Other Comments
This issue blocks

a-stark, 2017-08-01:
I will have a look into that.

drogenlied, 2017-08-02:
Simple test case, run from Qudi folder and comment SpinBox / uncomment ScienDSpinBox to reproduce fast.

```
# -*- coding: utf-8 -*-

import sys
from qtpy import QtCore, QtWidgets, QtGui
import pyqtgraph as pg
#from qtwidgets.scientific_spinbox import ScienDSpinBox

def main():
    
    app = QtWidgets.QApplication(sys.argv)

    w = QtWidgets.QWidget()
    #s = ScienDSpinBox()
    s = pg.SpinBox(int=False, suffix='Hz', siPrefix=True, dec=True, step=0.1, bounds=(0.0, 99.99))
    l = QtWidgets.QVBoxLayout()
    w.resize(250, 150)
    w.move(300, 300)
    w.setWindowTitle('Simple')
    w.setLayout(l)
    l.addWidget(s)
    w.show()
    s.editingFinished.connect(ed_finish)
    
    sys.exit(app.exec_())

def ed_finish():
    print('lol')

if __name__ == '__main__':
    main()

```

a-stark, 2017-08-03:
I tested your provided code, and I encountered no problems with that. Could you provide your python and pyqtgraph version? I have tested it for python 3.6.0 and python 3.6.2 with pyqtgraph.__version__ = 0.10.0 (I updated to the recent pyqtgraph version).

drogenlied, 2017-08-04:
You have to install pyqtgraph from the github repository.

a-stark, 2017-08-10:
Thanks for the note, I somehow skipped the information about using pyqtgraph-qo.
I would not recommend to use the pyqtgraph 10.0.2 version at the moment (which is the current pyqtgraph development branch). There seem to be some serious bugs causing an insane memory leak.

To this issue: the problem originates in the 'interpret' method of /qtwidgets/pyqtgraphmod/SpinBox.py in the line 'val = fn.siEval(strn)'. The function 'fn.siEval' has in the development version a different behaviour, i.e. eventually a different return type (here: str), causing the problem. The 'interpret' method has to be rewritten by using fn.siParse and fn.siApply (compare with the current version of [SpinBox.py](https://github.com/pyqtgraph/pyqtgraph/blob/develop/pyqtgraph/widgets/SpinBox.py) in the pygtgraph development branch). Moreover there are some significant changes, which should also be incorporated.

To the memory leak:
I encountered that the 'fn.siApply' method occupies my complete RAM, if large values are set. Here the modified test scenario from above:

```
# -*- coding: utf-8 -*-

import sys
from qtpy import QtCore, QtWidgets, QtGui
import pyqtgraph as pg
from qtwidgets.scientific_spinbox import ScienDSpinBox

def main():
    
    app = QtWidgets.QApplication(sys.argv)

    w = QtWidgets.QWidget()
    s = ScienDSpinBox()
    s.setMaximum(1e12)
    s.setValue(4e9)
    s.setSuffix('V')
    #s = pg.SpinBox(int=False, suffix='Hz', siPrefix=True, dec=True, step=0.1, bounds=(0.0, 99.99))
    l = QtWidgets.QVBoxLayout()
    w.resize(250, 150)
    w.move(300, 300)
    w.setWindowTitle('Simple')
    w.setLayout(l)
    l.addWidget(s)
    w.show()
     
    strn = s.lineEdit().text()
    val, siprefix, suffix = pg.fn.siParse(strn)
    value = pg.fn.siApply(val, siprefix)
    print('value:',value)    
    
    sys.exit(app.exec_())

if __name__ == '__main__':
    main()
```

kay-jahnke, 2018-03-22:
solved by Pull Request 342

commit c80144a2 (2016-08-11): close figures after saving, hopefully fixes #238
closed 2018-03-22 by kay-jahnke