Ulm-IQO/qudi-legacy #260: Matplotlib crashes Qudi upon restart

a-stark, 2017-09-11:
## *What* is affected by this bug?

I assume that this crash is mostly related to windows and it is connected to the usage of matplotlib. It is not possible any more to restart Qudi, if a graph was saved once by using matplotlib.

This seems to be closely related to issue #114.

## *When* does this occur?

If a figure is plotted and shown with the package matplotlib.

## *Where* on the platform does it happen?

Test where done on two different PCs:

PC nr 1: Windows 7 Pro, 64bit, Python 3.6.1
PC nr 2: Windows 7 Pro, 64bit, Python 3.6.2

## *How* do we replicate the issue?

In the manager console, type in the following:

```
import matplotlib.pyplot as plt
plt.figure()
plt.plot(1,2)
plt.show()           # after that the QCoreApplication message in the log appears.
```
Just to be sure, close all graphs manually:
```
plt.close('all')
```
and now restart Qudi.
Qudi is quitting almost perfectly but I assume just before a new QApplication (main event loop) is started Qudi crashes.

This is the output of the interpreter:
```
Used Qt API: PyQt5
Used Qt API: PyQt5
Loading Qudi...
============= Starting Manager configuration from <Path-to-Qudi>/qudi/config/iqo-config/alex_pc/custom_alex.cfg =================

============= Manager configuration complete =================

QCoreApplication::exec: The event loop is already running  <== this happens after plt.show()
Closing windows...
Stopping threads...

  Qudi is closed!  Ciao.
Unexpected return value 3221226519. Exiting.

Process finished with exit code -1
```

And that is the content of the log files:

```
2017-09-11 15:28:25 info __main__ Loading Qudi...
2017-09-11 15:28:27 debug core.manager Main thread is <sip.voidptr object at 0x00000000080D8F58>
2017-09-11 15:28:27 info core.manager load.cfg config file found at C:\Users\astark\Dropbox\Doctorwork\Software\QuDi-Git\qudi\config\load.cfg
2017-09-11 15:28:27 info core.manager Starting Manager configuration from C:/Users/astark/Dropbox/Doctorwork/Software/QuDi-Git/qudi/config/iqo-config/alex_pc/custom_alex.cfg
2017-09-11 15:28:27 info core.manager Manager configuration complete.
2017-09-11 15:28:27 info core.manager Qudi started.
2017-09-11 15:28:27 info core.manager Loading module ".gui.manager.managergui"
2017-09-11 15:28:28 info core.manager Configuring ManagerGui as man
2017-09-11 15:28:28 debug gui.manager.managergui.ManagerGui Activation in thread <sip.voidptr object at 0x000000000848A0A8>
2017-09-11 15:28:29 debug gui.manager.managergui.ManagerGui IPy activation in thread <sip.voidptr object at 0x0000000008615990>
2017-09-11 15:28:29 info gui.manager.managergui.ManagerGui IPython has kernel True
2017-09-11 15:28:29 info gui.manager.managergui.ManagerGui IPython kernel alive True
2017-09-11 15:28:30 debug core.manager Activation success: True
2017-09-11 15:28:30 info core.manager Loading module ".gui.trayicon"
2017-09-11 15:28:30 info core.manager Configuring TrayIcon as tray
2017-09-11 15:28:30 debug gui.trayicon.TrayIcon Activation in thread <sip.voidptr object at 0x000000000893DF08>
2017-09-11 15:28:30 debug core.manager Activation success: True
2017-09-11 15:28:30 info core.manager Loading module ".logic.taskrunner"
2017-09-11 15:28:30 info core.manager Configuring TaskRunner as tasklogic
2017-09-11 15:28:30 debug core.threadmanager Creating thread: "mod-logic-tasklogic".
2017-09-11 15:28:30 debug logic.taskrunner.TaskRunner Activation in thread <sip.voidptr object at 0x0000000008476FD0>
2017-09-11 15:28:30 info core.manager Task runner registered.
2017-09-11 15:28:30 debug core.manager Activation success: True
2017-09-11 15:29:01 warning Qt QCoreApplication::exec: The event loop is already running
2017-09-11 15:29:24 debug core.manager Deactivating ['man', 'tray', 'tasklogic']
2017-09-11 15:29:24 info core.manager Deactivating module logic.tasklogic
2017-09-11 15:29:24 info core.manager Deactivating logic.tasklogic
2017-09-11 15:29:24 debug logic.taskrunner.TaskRunner Deactivation in thread <sip.voidptr object at 0x0000000003929418>
2017-09-11 15:29:24 info core.manager Task runner removed.
2017-09-11 15:29:24 debug core.threadmanager Quitting thread mod-logic-tasklogic.
2017-09-11 15:29:24 debug core.threadmanager Waiting for thread mod-logic-tasklogic to end.
2017-09-11 15:29:24 debug core.manager Deactivation success: True
2017-09-11 15:29:24 debug core.threadmanager Thread mod-logic-tasklogic has quit.
2017-09-11 15:29:24 debug core.threadmanager Cleaning up thread mod-logic-tasklogic.
2017-09-11 15:29:24 info core.manager Deactivating module gui.tray
2017-09-11 15:29:24 info core.manager Deactivating gui.tray
2017-09-11 15:29:24 debug gui.trayicon.TrayIcon Deactivation in thread <sip.voidptr object at 0x000000000893DF08>
2017-09-11 15:29:24 debug core.manager Deactivation success: True
2017-09-11 15:29:24 info core.manager Deactivating module gui.man
2017-09-11 15:29:24 info core.manager Deactivating gui.man
2017-09-11 15:29:24 debug gui.manager.managergui.ManagerGui Deactivation in thread <sip.voidptr object at 0x000000000893DF08>
2017-09-11 15:29:24 debug gui.manager.managergui.ManagerGui IPy deactivation: <sip.voidptr object at 0x0000000006E47288>
2017-09-11 15:29:24 debug core.manager Deactivation success: True
2017-09-11 15:29:24 info __main__ Closing windows...
2017-09-11 15:29:24 info __main__ Stopping threads...
2017-09-11 15:29:24 debug core.threadmanager Quit all threads.
2017-09-11 15:29:24 info __main__ Qudi is closed!  Ciao.
```

I do not see anything suspicious except those message about the QCoreApplication::exec:.

Can you reproduce that on a windows maschine? I assume that will not happen on a unix os.

## Expected behavior (i.e. solution)

Maybe we have to specify a proper backend for matplotlib?

Can you reproduce this on other windows maschines (possibly on unix ones)?
Has some of you a clue where to start searching to debug and fix that?

tobiasgehring, 2017-09-11:
no problem on pyqt 5.9 pimped Fedora 26

tobiasgehring, 2017-09-11:
and I can reproduce it on Windows 7, 64 bit, python 3.6.2, Qt 5.9,
current pyqt5

a-stark, 2018-02-26:
This should be fixed with #326.

commit 5bdc7904 (2017-01-17): fix #260
closed 2018-02-26 by a-stark