Ulm-IQO/qudi-legacy #287: Oversampling for counter bug

SimonSchmitt88, 2017-10-23:
## *What* is affected by this bug?

Functionality of the slow counter on master branch. Problem is probably in counter_logic. More specific when changing the oversampling to >1 the following problems arise:

* after stopping the count frequency and the oversampling cannot be adjusted in the GUI
* when trying to record counts the counter throws an error:

[image]

[image]

(deleting ', 0' seems to help)
* this error prevents that the counter works after a restart (even for oversampling=1)
* the pdf and png of the recoreded data look very strange

## *When* does this occur?

Setting oversampling > 1 and trying ti record data. Probably the lines of code from above have to be changed

## *How* do we replicate the issue?

1. Load counter GUI
2. Change oversampling
3. Start counter
4. Record data

## Expected behavior (i.e. solution)

There should be no error message and the saved data should be the measured one

##Other Comments

LukePiewalker, 2020-10-29:
Since the slow counter module is now deprecated, I think this Issue can be closed right @Neverhorst?

Neverhorst, 2020-10-29:
That's right @LukePiewalker. Old `CounterLogic` has been completely replaced by `TimeSeriesReaderLogic`.

closed 2020-10-29 by Neverhorst