Ulm-IQO/qudi-legacy #289: Local config directory should be more generally named

latchr, 2017-10-24:
## *What* is affected by this bug?
The "qo-config" local config dir is hard-listed in the main project .gitignore file. Other users may not want their own config in a directory with this name.

## *When* does this occur?
As soon as the repo is cloned, the .gitignore file forces "qo-config" naming.

## *Where* on the platform does it happen?
.gitignore file

## *How* do we replicate the issue?
Look in .gitignore

## Expected behavior (i.e. solution)
I suggest the ignored local config dir gets a name change to "local". Then users can simply put config files in config/local/ or groups can have their own subdirectories such as config/local/qo-config/

##Other Comments

This change is important to achieve the general usability that we desire. However it will require some work in groups that already use Qudi - because the local config dir will need to be renamed manually in every case.

latchr, 2017-10-24:
Perhaps a good solution is to add a "config/local" entry to the .gitignore file now (easy and doesn't effect anyone), and then allow a period of time for users to migrate local config across to the new dir name.

latchr, 2017-11-02:
I have thought this over for a week, and there has not been a lot of other suggestions. I propose adding "/config/local" to the .gitignore file (and leaving the "/config/qo-config" for now). This will not break anyone at Ulm-IQO, but will make it nicer for users elsewhere who want to have local configs.

@drogenlied @a-stark @jochen-scheuer @Neverhorst do you have any objections to this solution?

a-stark, 2017-11-02:
Hi @latchr, maybe I got the problem not completely, but I encounter so far no problems here. 
My config files are for instance all in the folder '/config/iqo-config/*' where 'iqo-config' is its one repository (note that 'iqo-config' not included in .gitignore). If you do not add files explicitly to your local qudi repository so that git tracks them, then git will simply ignore them. In the other case, if you want to put your config file in a separate repository (like I have it described above), then there should be also no problem in including this folder to the '/config' folder.  
Please correct me if I miss here something.

tobiasgehring, 2017-11-02:
BTW: why would you want to store config files *in* the qudi software directory at all? I think this is bad style. Imagine qudi is installed under /usr/, then you couldn't store anything in its directory. If you want general usability, you should not advise people to store their config and also local modules in the qudi directory!

a-stark, 2017-11-02:
that is a very valid point and the functionality is already present to select a config being situated elsewhere on the system.

latchr, 2017-11-02:
@a-stark, What happens when you go to your qudi root directory and type "git status". Surely if you have directories in config that are not listed in the .gitignore, then they must show up as untracked files. For me, with a "local" directory in /config I get the following:

```
Untracked files:
  (use "git add <file>..." to include in what will be committed)

        config/local/
```
If the user is familiar with git and is happy to manually ignore this message, then everything is fine. However, some tools (like TortoiseGit from memory) automatically like to add new untracked files. This would result in commits that put local user configs into the qudi repository.

latchr, 2017-11-02:
@tobiasgehring I see exactly what you are saying, and this will be even more important when qudi is an installable python package.  However, at the moment the qudi install instructions outline a process for getting the full development tool-chain. In this situation, the traditional Qudi usage is to have local user config stored in a subdirectory of /config in the qudi project root.

You can clearly see this "tradition" by the fact that /config/qo-config is listed in the main repository .gitignore file.

I am claiming that it makes far more sense for this hard-coded local config dir to be renamed to something more general than "qo-config" (which has meaning in the Ulm institute for Quantum Optics, but is bizarrely arbitrary elsewhere in the world).

I therefore see two options:

1. We add a "/config/local" directory to the .gitignore file and make the documentation clearly suggest this as a location for local config files. After a transition period in which the Ulm-IQO group can move all their config subdirectories to /config/local it would then be best to remove the "config/qo-config" listing from the .gitignore file.

2. We remove all local config directories from the .gitignore and clearly instruct people to curate their local Qudi configs elsewhere on their computer.  This still requires Ulm-IQO to do some significant changes in each of their deployed qudi instances.

I am strongly in favour of option 1, because I see this as maintaining the "lowest barrier to entry" for a not-thoroughly-technical user.  From my interactions with students and colleagues, I am sure that having config files for Qudi held with the Qudi "folder" makes sense for lots of people.  As one simple example, lots of Qudi users work in an IDE (like PyCharm) that gives a directory-tree of the project. It is regularly necessary to edit code and edit config roughly at the same time. Having local config files in the directory tree of the Qudi project is by far the most "obvious" way to handle quick browsing back and forth. This is precisely why it has traditionally been done this way in Qudi.

a-stark, 2017-11-02:
Well sure, I got of course a similar output for `git status`, but to be correct, TortoiseGit will not add untracked files automatically to your repository. It is true that they will be listed as 'Not Versioned Files' in a separate category, but the user needs actively to select these files. (See for instance my current screenshot):

[image]

But in essence, I am not opposed to option 1 or 2. I also do not see any problems in adding '\config\local\*' to `.gitignore`. If you make a pull request for that, it would be great if you update also the installation procedure suggesting how to handle local configs.

liuyanulm, 2020-10-09:
I have followed the conversation here. It looks to me that the issue has been resolved with a merged PR. and I don't see 'qo-config' in the current qudi version. So, I am closing the PR.

closed 2020-10-09 by liuyanulm

Neverhorst, 2020-10-10:
New qudi core will be using AppData and/or user home directory for app status, config, measurement data etc.