Ulm-IQO/qudi-legacy #327: Windows not showing up after firing up the GUI

hellbourne, 2018-02-22:
## *What* is affected by this bug?

The entire qudi

## *When* does this occur?

When I start by running 'python start.py' or from pycharm

## *Where* on the platform does it happen?

Windows 10. New computer. On the old computer it runs OK.

## *How* do we replicate the issue?

I see the diamond icon on the taskbar at the bottom of the screen but I cannot open the main window nor can I see it. If I hover over the icon I do see the miniature of the window

## Expected behavior (i.e. solution)

The window should have appreared

## Other Comments

a-stark, 2018-02-22:
I suppose that the installation instructions were not followed correctly. On Windows 10, you need to install Anaconda for **all users** (which requires admin privileges). Could you repeat the anaconda installation and then also the installation of Qudi described [in the docs](privileges)?

hellbourne, 2018-02-22:
Solved by pulling the newest version.

a-stark, 2018-02-22:
awesome.

closed 2018-02-22 by a-stark