Ulm-IQO/qudi-legacy #354: Problems with the fit accuracy of period

SimonSchmitt88, 2018-03-29:
## *What* is affected by this bug?

When doing any sine-fit the fit 'error' of the period sometimes seems to be too large. (Not sure if this true  for other fit-function as well)

## *When* does this occur?

In the example data below, it is not possible to obtain the period from the fitparameters due to the large error, which seems although the data is noisy and not even a full period is fitted, too large. Also the relative error in the frequency estimate is much lower, what seems surprising.

[image]

[image]

## *Where* on the platform does it happen?

## *How* do we replicate the issue?

## Expected behavior (i.e. solution)

The error of the period should be smaller and the relative errors of period and frequency should be the same.

## Other Comments

SimonSchmitt88, 2018-03-30:
Thanks! I think this is exactly the problem and frequency_error/(frequency**2) should be the correct equation. I will test your branch as soon as possible.

Do you mean with contrast just twice the amplitude or something more specific?

Neverhorst, 2018-06-04:
Fixed by PR #370

closed 2018-06-04 by Neverhorst