Ulm-IQO/qudi-legacy #355: fastcomtec np.int32 error with long data files

jochen-scheuer, 2018-03-30:
## *What* is affected by this bug?

Pull and analyze does not work when file is too long

## *When* does this occur?

When the length of the array pulled from fastcomtec is longer than np.int32 .

[image]

_get_exception_class.<locals>.Derived: 

========= Remote Traceback (1) =========
Traceback (most recent call last):
  File "C:\Anaconda3\envs\qudi\lib\site-packages\rpyc\core\protocol.py", line 305, in _dispatch_request
    res = self._HANDLERS[handler](self, *args)
  File "C:\Anaconda3\envs\qudi\lib\site-packages\rpyc\core\protocol.py", line 535, in _handle_call
    return self._local_objects[oid](*args, **dict(kwargs))
  File "D:\software\qudi-git\hardware\fastcomtec\fastcomtecp7887.py", line 341, in get_data_trace
    data = np.empty((N,), dtype=np.uint32)
MemoryError

Traceback (most recent call last):

  File "C:\Users\Confocal\Documents\qudi-git\logic\pulsed_measurement_logic.py", line 726, in _pulsed_analysis_loop
    fc_data = netobtain(self._fast_counter_device.get_data_trace())

  File "C:\Anaconda3\envs\qudi\lib\site-packages\rpyc\core\netref.py", line 196, in __call__
    return syncreq(_self, consts.HANDLE_CALL, args, kwargs)

  File "C:\Anaconda3\envs\qudi\lib\site-packages\rpyc\core\netref.py", line 71, in syncreq
    return conn.sync_request(handler, oid, *args)

  File "C:\Anaconda3\envs\qudi\lib\site-packages\rpyc\core\protocol.py", line 441, in sync_request
    raise obj

## *Where* on the platform does it happen?

Windows 
## *How* do we replicate the issue?

## Expected behavior (i.e. solution)

Not depending on file length it should be possible to do pull and analyze

## Other Comments

[20180330_numpyint64error.zip](https://github.com/Ulm-IQO/qudi/files/1863488/20180330_numpyint64error.zip)

drogenlied, 2018-04-03:
`MemoryError` means that the Python interpreter failed to reserve the desired amount of memory from the operating system.

On 32bit operating systems, the hard upper limit for a single process is 4GB.
Windows and Linux used to split this memory address space into 3GB user/1GB system spaces (but configurable to e.g. 2/2GB) for performance reasons.
(Spectre/Meltdown fixes change that, there should be no more split).

For this reason, the per process upper limit can also be 2GB or 3GB and there is nothing that can be done about this at runtime.

If you are on a 64bit system, you are probably out of memory.

Neverhorst, 2018-10-25:
Is that still an issue? Or has it been a one time incident because the trace data from the fastcomtec device was simply too long to be stored in the PC memory as 64bit integer array?
It would be great if you could comment on that, @jochen-scheuer.

Neverhorst, 2019-05-16:
Lack of interest by author...

closed 2019-05-16 by Neverhorst