Ulm-IQO/qudi-legacy #399: on SMIQ, rf output cable vibrates, when running a list if cw power is different with list mode.

liuyanulm, 2018-08-30:
## *What* is affected by this bug?

A tiny sample drift could happen.

## *Why* does this occur?

Because in mw_source_smiq.py, list_on would run when a cw_on is performed. see the function below
    def list_on(self):
        """
        Switches on the list mode microwave output.
        Must return AFTER the device is actually running.

        @return int: error code (0:OK, -1:error)
        """
        current_mode, is_running = self.get_status()
        if is_running:
            if current_mode == 'list':
                return 0
            else:
                self.off()

        # This needs to be done due to stupid design of the list mode (sweep is better)
        self.cw_on()
        self._command_wait(':LIST:LEARN')
        self._command_wait(':FREQ:MODE LIST')
        dummy, is_running = self.get_status()
        while not is_running:
            time.sleep(0.2)
            dummy, is_running = self.get_status()
        return 0
I am not exactly sure about the reason. probably with internal attenuator swap stuff.

## *How* to fix the issue?

set cw power to be the same as list power, by adding this       
self.set_cw(2870e9,float('{0}'.format(self.get_power())))
before
self.cw_on()

liuyanulm, 2020-10-07:
This happens in very old SMIQ. not reproduced on not so old SMIQs.
I am not sure if this is necessary any more.
@NicolasStaudenmaier

NicolasStaudenmaier, 2020-10-07:
If this happens for old devices and cannot be reproduced anymore I think that it's a very individual device specific issue (maybe the SMIQ has some defects) and not regarding the software. In that case we can close the issue.

liuyanulm, 2020-10-08:
Since no one else has been bothered for this issue. I close it.

closed 2020-10-08 by liuyanulm