Ulm-IQO/qudi-legacy #431: Spincore pulse blaster hardware not working

alrik-durand, 2018-10-04:
The hardware module 'spincore.pulse_blaster_esrpro' states that it implement the PulserInterface, but it does not.
I tried implementing it by myself but I'm quite lost on pulsed measurement as it is very complex and hard to get into. 
Can Qudi work only with sequence mode only ? Or waveform is needed ?

If someone here can head me in the right direction I would greatly appreciate it :)

a-stark, 2018-10-04:
in fact, it is already implemented, tested and working with Qudi (with the pulsed v3 update), just sitting in another branch, [pulse_blaster_implementation](https://github.com/Ulm-IQO/qudi/tree/pulse_blaster_implementation). I need to include the latest pulsed modifications for it and test it once again.

alrik-durand, 2018-10-04:
Well you just made my day a lot better ! I'll have a look at it.

Neverhorst, 2018-10-10:
Fixed by PR #432

closed 2018-10-10 by Neverhorst