Ulm-IQO/qudi-legacy #463: Remote access: error while importing matplotlib.pyplot

ps-lukin-siv, 2018-12-15:
## *What* is affected by this bug?
Remoteload of any module, containing _import matplotlib.pyplot_.
Examples:
- It happens at start-up of Qudi Jupyter Notebook
- remoteload of a minimal hardware module with _import matplotlib.pyplot_ in the very beginning

## *When* does this occur?
During "remoteload" of a module with _import matplotlib.pyplot_

## *Traceback*
In the two examples below same sequence of steps happens after _import matplotlib.pyplot_
and the finial error message is:
_RuntimeError: There is no current event loop in thread ..._

**Qudi Jupyter Notebook example:**
Traceback (most recent call last):
  File "C:\Users\Roger\qudi\logic\jupyterkernel\qzmqkernel.py", line 183, in connect_kernel
    self.init_exec_env()
  File "C:\Users\Roger\qudi\logic\jupyterkernel\qzmqkernel.py", line 196, in init_exec_env
    setup_matplotlib(self)
  File "C:\Users\Roger\qudi\logic\jupyterkernel\helpers.py", line 201, in setup_matplotlib
    **import matplotlib.pyplot**
  File "C:\Users\Roger\Anaconda3\envs\qudi\lib\site-packages\matplotlib\pyplot.py", line 2609, in <module>
    install_repl_displayhook()
  File "C:\Users\Roger\Anaconda3\envs\qudi\lib\site-packages\matplotlib\pyplot.py", line 166, in install_repl_displayhook
    ip.enable_gui(ipython_gui_name)
  File "C:\Users\Roger\Anaconda3\envs\qudi\lib\site-packages\ipykernel\inprocess\ipkernel.py", line 175, in enable_gui
    enable_gui(gui, kernel=self.kernel)
  File "C:\Users\Roger\Anaconda3\envs\qudi\lib\site-packages\ipykernel\eventloops.py", line 374, in enable_gui
    kernel.eventloop = loop
  File "C:\Users\Roger\Anaconda3\envs\qudi\lib\site-packages\traitlets\traitlets.py", line 585, in __set__
    self.set(obj, value)
  File "C:\Users\Roger\Anaconda3\envs\qudi\lib\site-packages\traitlets\traitlets.py", line 574, in set
    obj._notify_trait(self.name, old_value, new_value)
  File "C:\Users\Roger\Anaconda3\envs\qudi\lib\site-packages\traitlets\traitlets.py", line 1139, in _notify_trait
    type='change',
  File "C:\Users\Roger\Anaconda3\envs\qudi\lib\site-packages\traitlets\traitlets.py", line 1176, in notify_change
    c(change)
  File "C:\Users\Roger\Anaconda3\envs\qudi\lib\site-packages\ipykernel\kernelbase.py", line 52, in _update_eventloop
    loop = ioloop.IOLoop.current()
  File "C:\Users\Roger\Anaconda3\envs\qudi\lib\site-packages\tornado\ioloop.py", line 282, in current
    loop = asyncio.get_event_loop()
  File "C:\Users\Roger\Anaconda3\envs\qudi\lib\asyncio\events.py", line 694, in get_event_loop
    return get_event_loop_policy().get_event_loop()
  File "C:\Users\Roger\Anaconda3\envs\qudi\lib\asyncio\events.py", line 602, in get_event_loop
    % threading.current_thread().name)
RuntimeError: There is no current event loop in thread 'Dummy-12'.
2018-12-14 18:32:39 info logic.jupyterkernel.kernellogic.QudiKernelLogic Finished starting Kernel 55d8a6ac-1b62-4fbe-a224-e99bc01a1efc

**Example of a trivial hardware module with _import matplotlib.pyplot_ line in the beginning of the file**
2018-12-14 18:39:54 info core.remote Client connected!
2018-12-14 18:39:54 info core.remote remotesearch: rngRoger
2018-12-14 18:39:54 info core.manager Loading module ".hardware.rng"
2018-12-14 18:39:55 error core.manager Error while loading hardware module: rngRoger
Traceback (most recent call last):
  File "C:\Users\Roger\qudi\core\manager.py", line 795, in loadConfigureModule
    modObj = self.importModule(base, module_name)
  File "C:\Users\Roger\qudi\core\manager.py", line 549, in importModule
    baseName, module), fromlist=['*'])
  File "<frozen importlib._bootstrap>", line 1074, in __import__
  File "<frozen importlib._bootstrap>", line 994, in _gcd_import
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 665, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 678, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "C:\Users\Roger\qudi\hardware\rng.py", line 4, in <module>
    **import matplotlib.pyplot as plt**
  File "C:\Users\Roger\Anaconda3\envs\qudi\lib\site-packages\matplotlib\pyplot.py", line 2609, in <module>
    install_repl_displayhook()
  File "C:\Users\Roger\Anaconda3\envs\qudi\lib\site-packages\matplotlib\pyplot.py", line 166, in install_repl_displayhook
    ip.enable_gui(ipython_gui_name)
  File "C:\Users\Roger\Anaconda3\envs\qudi\lib\site-packages\ipykernel\inprocess\ipkernel.py", line 175, in enable_gui
    enable_gui(gui, kernel=self.kernel)
  File "C:\Users\Roger\Anaconda3\envs\qudi\lib\site-packages\ipykernel\eventloops.py", line 374, in enable_gui
    kernel.eventloop = loop
  File "C:\Users\Roger\Anaconda3\envs\qudi\lib\site-packages\traitlets\traitlets.py", line 585, in __set__
    self.set(obj, value)
  File "C:\Users\Roger\Anaconda3\envs\qudi\lib\site-packages\traitlets\traitlets.py", line 574, in set
    obj._notify_trait(self.name, old_value, new_value)
  File "C:\Users\Roger\Anaconda3\envs\qudi\lib\site-packages\traitlets\traitlets.py", line 1139, in _notify_trait
    type='change',
  File "C:\Users\Roger\Anaconda3\envs\qudi\lib\site-packages\traitlets\traitlets.py", line 1176, in notify_change
    c(change)
  File "C:\Users\Roger\Anaconda3\envs\qudi\lib\site-packages\ipykernel\kernelbase.py", line 52, in _update_eventloop
    loop = ioloop.IOLoop.current()
  File "C:\Users\Roger\Anaconda3\envs\qudi\lib\site-packages\tornado\ioloop.py", line 282, in current
    loop = asyncio.get_event_loop()
  File "C:\Users\Roger\Anaconda3\envs\qudi\lib\asyncio\events.py", line 694, in get_event_loop
    return get_event_loop_policy().get_event_loop()
  File "C:\Users\Roger\Anaconda3\envs\qudi\lib\asyncio\events.py", line 602, in get_event_loop
    % threading.current_thread().name)
RuntimeError: There is no current event loop in thread 'Thread-10'.
2018-12-14 18:39:55 warning core.manager Stopping module loading after loading failure.

## Other Comments
Loading of a local module with _import matplotlib.pyplot_ works well. The error also does not appear if you manually execute _import matplotlib.pyplot_ in the console before initiating remoteload.

A naive suspicion would be that host's (server's) _install_repl_displayhook()_ function of pyplot.py somehow confuses client's and host's IPythons and tries to perform _ip.enable_gui(ipython_gui_name)_ using the client's _ipython_gui_name_.

ps-lukin-siv, 2018-12-18:
The same problem happens at 3 different computers:
**PC1**
_OS:_ Windows 10 (64 bit)
_Conda version:_
```
(qudi) C:\Users\Roger\Anaconda3\envs>conda info

     active environment : qudi
    active env location : C:\Users\Roger\Anaconda3\envs\qudi
            shell level : 1
       user config file : C:\Users\Roger\.condarc
 populated config files : C:\Users\Roger\.condarc
          conda version : 4.5.11
    conda-build version : 3.10.5
         python version : 3.6.5.final.0
       base environment : C:\Users\Roger\Anaconda3  (writable)
           channel URLs : https://repo.anaconda.com/pkgs/main/win-64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/free/win-64
                          https://repo.anaconda.com/pkgs/free/noarch
                          https://repo.anaconda.com/pkgs/r/win-64
                          https://repo.anaconda.com/pkgs/r/noarch
                          https://repo.anaconda.com/pkgs/pro/win-64
                          https://repo.anaconda.com/pkgs/pro/noarch
                          https://repo.anaconda.com/pkgs/msys2/win-64
                          https://repo.anaconda.com/pkgs/msys2/noarch
          package cache : C:\Users\Roger\Anaconda3\pkgs
                          C:\Users\Roger\AppData\Local\conda\conda\pkgs
       envs directories : C:\Users\Roger\Anaconda3\envs
                          C:\Users\Roger\AppData\Local\conda\conda\envs
                          C:\Users\Roger\.conda\envs
               platform : win-64
             user-agent : conda/4.5.11 requests/2.18.4 CPython/3.6.5 Windows/10 Windows/10.0.17134
          administrator : False
             netrc file : None
           offline mode : False
```

**PC2**
_OS:_ Windows 10 (64 bit)
_Conda version:_
```
(qudi) C:\Windows\System32>conda info

     active environment : qudi
    active env location : C:\Anaconda\envs\qudi
            shell level : 1
       user config file : C:\Users\Lukin SiV\.condarc
 populated config files : C:\Users\Lukin SiV\.condarc
          conda version : 4.5.11
    conda-build version : 3.15.1
         python version : 3.7.0.final.0
       base environment : C:\Anaconda  (writable)
           channel URLs : https://repo.anaconda.com/pkgs/main/win-64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/free/win-64
                          https://repo.anaconda.com/pkgs/free/noarch
                          https://repo.anaconda.com/pkgs/r/win-64
                          https://repo.anaconda.com/pkgs/r/noarch
                          https://repo.anaconda.com/pkgs/pro/win-64
                          https://repo.anaconda.com/pkgs/pro/noarch
                          https://repo.anaconda.com/pkgs/msys2/win-64
                          https://repo.anaconda.com/pkgs/msys2/noarch
          package cache : C:\Anaconda\pkgs
                          C:\Users\Lukin SiV\AppData\Local\conda\conda\pkgs
       envs directories : C:\Anaconda\envs
                          C:\Users\Lukin SiV\AppData\Local\conda\conda\envs
                          C:\Users\Lukin SiV\.conda\envs
               platform : win-64
             user-agent : conda/4.5.11 requests/2.19.1 CPython/3.7.0 Windows/10 Windows/10.0.15063
          administrator : False
             netrc file : None
           offline mode : False
```

**PC3**
_OS:_ Windows 7 (64 bit)
_Conda version:_
```
(qudi) C:\Users\Alp>conda info

     active environment : qudi
    active env location : C:\Users\Alp\Anaconda3\envs\qudi
            shell level : 1
       user config file : C:\Users\Alp\.condarc
 populated config files : C:\Users\Alp\.condarc
          conda version : 4.5.11
    conda-build version : 3.15.1
         python version : 3.7.0.final.0
       base environment : C:\Users\Alp\Anaconda3  (writable)
           channel URLs : https://conda.anaconda.org/conda-forge/win-64
                          https://conda.anaconda.org/conda-forge/noarch
                          https://repo.anaconda.com/pkgs/main/win-64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/free/win-64
                          https://repo.anaconda.com/pkgs/free/noarch
                          https://repo.anaconda.com/pkgs/r/win-64
                          https://repo.anaconda.com/pkgs/r/noarch
                          https://repo.anaconda.com/pkgs/pro/win-64
                          https://repo.anaconda.com/pkgs/pro/noarch
                          https://repo.anaconda.com/pkgs/msys2/win-64
                          https://repo.anaconda.com/pkgs/msys2/noarch
          package cache : C:\Users\Alp\Anaconda3\pkgs
                          C:\Users\Alp\AppData\Local\conda\conda\pkgs
       envs directories : C:\Users\Alp\Anaconda3\envs
                          C:\Users\Alp\AppData\Local\conda\conda\envs
                          C:\Users\Alp\.conda\envs
               platform : win-64
             user-agent : conda/4.5.11 requests/2.19.1 CPython/3.7.0 Windows/7 W
indows/6.1.7601
          administrator : True
             netrc file : None
           offline mode : False
```

drogenlied, 2018-12-19:
It seems a little strange that two of them have Python 3.7.0 as their Python version.

How did you install the conda environment?

Can you post the output of `conda env export` for the qudi environment?

ps-lukin-siv, 2018-12-22:
Jan, apologize for the delayed response.

_How did you install the conda environment?_
**PC1:** I've just installed Anaconda distribution: `https://repo.anaconda.com/archive/Anaconda3-5.2.0-Windows-x86_64.exe`

**PC2**  and **PC3**. They had some Anaconda initially. After removing them (don't remember why), installed the following Anaconda distributions
PC2: `https://repo.anaconda.com/archive/Anaconda3-5.3.1-Windows-x86_64.exe`
PC3: `https://repo.anaconda.com/archive/Anaconda3-5.3.1-Windows-x86_64.exe` - not sure, can't find the downloaded installer

_Can you post the output of conda env export for the qudi environments?_
The outputs of `conda env export` for PC1 and PC2 are identical to the original file 
`qudi/tools/conda-env-win10-64bit-qt5.yml`
(`https://github.com/Ulm-IQO/qudi/blob/master/tools/conda-env-win10-64bit-qt5.yml`) 

As for PC3, the output is identical except for one extra channel:
```
channels:
  - conda-forge
  - defaults
```

_Can you try downgrading the tornado package to version 4.5.3?_
Will do this a bit later.

ps-lukin-siv, 2018-12-22:
Downgrading tornado to 4.5.3 fixed the problem! Now Jupyter Notebook and other modules are remoteloaded successfully! Thank you so much, Jan!

May be it makes sense to replace `tornado=5.0.2=py36_0` with `tornado=4.5.3=py36_0` in the 
environment build file (https://github.com/Ulm-IQO/qudi/blob/master/tools/conda-env-win10-64bit-qt5.yml).

Seems that tornado 5.0 indeed has some breaking changes as compared to 4.5.3. There is some discussion of similar issues (https://github.com/jupyter/notebook/issues/3407).

drogenlied, 2018-12-23:
Ok, I will make sure that all environments have tornado version 4.5.3.

PR #467 (2019-01-01): Always install tornado 4.5.3 to avoid problems with matplotlib

## Motivation and Context

Fixes #463.

## How Has This Been Tested?

Needs testing on Win7 and Win10, all other environments had the right version from the start.
Reinstall the environment and run one of the problematic cases from #463.

## Types of changes

- [x] Bug fix (non-breaking change which fixes an issue)
- [ ] New feature (non-breaking change which adds functionality)
- [ ] Breaking change (fix or feature that would cause existing functionality to not work as expected)

## Checklist:

- [ ] My code follows the code styl [... cut]
closed 2019-01-01 by drogenlied