Ulm-IQO/qudi-legacy #496: Need "skimage.morphology.local_maxima()" to find all fluorescent spots at once

diamond2nv, 2019-03-28:
## *What* is affected by this bug?

- [x] qudi\logic\confocal_logic.py
- [x] Confocal Gui
- [x] Automation Gui

## Expected behavior (i.e. solution)
- [ ] List all fluorescent spots in confocal-scan-image for automation

## Other Comments
a little test run:
```python
import numpy as np
import matplotlib.pyplot as plt
#
import skimage.morphology as morph
import skimage.filters as skif

#Create a floating point grayscale
img = Z0 / Z0.max()
    
# Global threshold
global_thresh = skif.threshold_otsu(img)
#global_thresh = 0.5
print(global_thresh)
    
#Coordinates and filters
lm = morph.local_maxima(img)
xc1, yc1 = np.where(lm.T == True)
v = img[(yc1, xc1)]
lim = global_thresh*2.0
if lim<1.0:
    xc2, yc2 = xc1[v > lim], yc1[v > lim]
    print('lim=',lim)
else:
    lim = global_thresh
    xc2, yc2 = xc1[v > lim], yc1[v > lim]
    print('lim=',lim)
```

diamond2nv, 2019-03-28:
test result:
[image]

https://github.com/diamond2nv/pyNVscan
Anaconda4，Python2.7.13, Scipy～0.19

liuyanulm, 2019-03-28:
Hi,
Thank you for bring up this issue with the method.
Perhaps it is more feasible to add this functionality into the poi_manager_logic in Qudi, instead of confocal_logic.
Cheers,
Yan Liu

diamond2nv, 2019-03-29:
yeah, you are right ! 
thanks very much for the help !

diamond2nv, 2019-04-06:
ref: auto_catch_poi functionality #501 by liuyanulm

https://github.com/Ulm-IQO/qudi/pull/501

[image]

diamond2nv, 2019-04-30:
```python
# Global threshold
global_thresh = skif.threshold_otsu(img)
#global_thresh = 0.5
```

- This is the threshold of float_count_value.

```python
#Coordinates and filters
lm = morph.local_maxima(img)
```

- This is the threshold of pixel value, most nearby (local_maxima(img) of xy scan image)

- When scan step size is large, global threshold of count value will became large?

and the threshold of pixel most nearby (of xy scan image) will became large too.