Ulm-IQO/qudi-legacy #503: Use official Physik Instrumente (PI) GCS python package -- PIPython to control most of their stages.

alex0130, 2019-04-16:
## *What* is affected by this issue?

- [ ]  qudi/hardware/motor/motor_stage_micos.py 

- [ ]  qudi/hardware/motor/motor_stage_pi.py 

## *Where* on the platform does it happen?

- [ ] on-board linux

## *How* do we replicate the issue?

- [ ] use moudle "PIPython" instead of visa-COM and visa-PCIe for Physik Instrumente (PI) GCS Devices.

- [ ] https://github.com/git-anonymous/PIPython

```python
from pipython import GCSDevice
pi_device = GCSDevice () 
# Load PI Python Libraries

pi_device.ConnectUSB ('123456789') 
# Connect to the controller via USB

pi_device.SVO ('A', 1) 
# Turn on servo control of axis "A"

pi_device.MOV ('A', 3.142) 
# Command axis  "A" to position 3.142

position = pi_device.qPOS ('A')
# Query current position of axis  "A"
```

## Expected behavior (i.e. solution)

- [ ] PIPython is a collection of Python modules to access a PI device and process GCS data. It can be used with Python 2.7+ and 3.4+ (3.6 is OK) on Windows, Linux and OS X and without the GCS DLL also on any other platform.

## Other Comments

pip install --upgrade git+https://github.com/git-anonymous/PIPython.git

diamond2nv, 2019-04-26:
@alex0130 
 * first,  should read the manual with your PI controller, make sure it has done need-to-do before run.
 * then,  can have a try test : https://github.com/diamond2nv/qudi/blob/POI_autocatch/hardware/motor/piezo_stage_pi_py_gcs2.py

Domx93, 2020-11-04:
Hello, I can connect the device (a E-870 piezo controller) easily, but then when I use for example pidevice.SVO ('A', 1)  it says "GCS Error: Unknown command (2)". If I use "pidevice.errcheck=False" it doesn't produce an error, but it doesn't do anything, and  pidevice.qPOS('A') produce an empty dictionary and pidevice.axes an empy array. Could you maye help me please?