Ulm-IQO/qudi-legacy #505: User and code documentation about Qudi shows no images

Baiyi-Yu, 2019-04-26:
## *What* is affected by this bug?

The website of user and code documentation about Qudi does not show any images.

## *When* does this occur?

## *Where* on the platform does it happen?

I'm using Chrome of latest version in Windows 10. And I've tried using edge or IE, but no help.

## *How* do we replicate the issue?

## Expected behavior (i.e. solution)

## Other Comments

Neverhorst, 2019-05-06:
Has anyone else encountered this problem? To my knowledge this is the first and only occurrence of this issue.
@Baiyi-Yu: Are you referring to https://ulm-iqo.github.io/qudi-generated-docs/html-docs/ ?

liuyanulm, 2019-05-16:
Are there actually any images in the documentation?
I have never seen any... But it does not matter to me as a user.

liuyanulm, 2020-10-07:
Hi @Lightup1, 
Could you please give some further description of your issue?
Is it that you do not have the access to the documentation, or you are not seeing any images in the documentation?
If it is the latter, please specify on which part of the documentation you suppose to see images while you could not see any.

Lightup1, 2020-10-07:
[image]
Hi @liuyanulm , 
the page is like this. 
Image can not be loaded.

liuyanulm, 2020-10-07:
> [image]
> Hi @liuyanulm ,
> the page is like this.
> Images can not be loaded.

OK I see. This is also reproduced on my side. Thank you for raising this issue.
We will fix this soon.