Ulm-IQO/qudi-legacy #507: Need enhancement to store large POIs' info into SQLite, MySQL and store Date files (images) into HDF5.

diamond2nv, 2019-04-26:
## Expected behavior (i.e. solution)

- [ ] store large POIs' information into SQLite, MySQL 
- [ ] store Date files (images) into HDF5 (backup ?)

## Other Comments

### SQL and HDF5 to support auto long-time experiment.

Neverhorst, 2019-04-29:
Please (at least roughly) stick to the issue template.
Please define more precisely what you mean.

diamond2nv, 2019-04-30:
## *What* is affected by this bug?

 * platform: Win 10 64bit; 
 * branch: master, 
 * branch: POI_autocatch,
 * modules: logic\poi_manager_logic.py

## *When* does this occur?

 * branch : master: POIs' information (name/tag, x ,y, y) is stored in lots files ,however, without URL combination in SQL, not good enough to manage lots of POIs for one bulk diamond sample(Chip/Wafer).
 * branch : POI_autocatch: "Auto Catch" for large scan step size.

## *Where* on the platform does it happen?

 * branch : POI_autocatch: "Auto Catch" will cause non-stopable behave.

## *How* do we replicate the issue?

1. Add sqlite (memory and disk) in "save_logic.py" for "poi_manager_logic.py"
    * ***Store POIs' information into SQLite***. And in SQL, create combination of all Data-files-URL with the POI info. 
    * ref: Zotero, Calibre...
2. Add some kind of interface.local_maxima for "logic\poi_manager_logic.py" and "logic\auto_catch_logic.py"(need be added too)
    * interface.local_maxima tack in confocal image from "poi_manager_logic.py", 
        * var: ndarry.scan_image( x_pixels_int, y_pixels_int, count_value_float), 
        * scanning_step_size_float_(m), 
        * tag_string, (like: "poi_20190430_auto")
        * and maybe a tunable s/n threshold for manual intervention;
        * and maybe a bash_string, (8 random chars) to istinguish the task/measurement for future use(like to do auto-catch by deep learning...***exchanging data in Redis***)
    * "auto_catch_logic.py" using "import skimage.morphology.local_maxima()" to auto discover all fluorescent spots in a scan image, this method used a lot in tracking stars in astronomy.
        * return (x_pixels_int, y_pixels_int, lim_threshold_float, poi_auto_name_full_string)
             * poi_auto_name_full_string: like "poi_20190430_auto_bashdfohls"
3.  ***Add HDF5 support in "save_logic.py" for all Data***. Can be support in OriginLab、Matlab, and this is good data/info managed as well.
## Expected behavior (i.e. solution)

 * add POI auto, fast
 * with data/info good managed in SQL for diamond samples.

## Other Comments

Neverhorst, 2019-05-06:
Mixing of several issues/requests.
Confusing description and no clear summary of what is meant.
This kind of "all-in-one" issues will not be tolerated. 
Please only submit issues regarding master.
Precisely define an issue/feature request for a single, broken-down topic.

closed 2019-05-06 by Neverhorst