Ulm-IQO/qudi-legacy #510: Bugs to be fix, need a little roll back of "confocalgui.py" and  "poimangui.py"

diamond2nv, 2019-05-05:
## *What* is affected by this bug?

Win10 64bit 
branch: master
modules: gui

## *When* does this occur?

1. When moving crosshair in Confocal GUI xy scan, the crosshair in depth xz is not moving accordingly( x-axis).
1. When using the tool "Add POIs" in POI Manager GUI, will ocur an error.

## *Where* on the platform does it happen?

- will be solved by a little roll back.

## *How* do we replicate the issue?

1. "confocalgui.py" -- roll back from line: 1089
below is from the same file with a few commits roll back:
```python
 #FIXME: little roll back
        if self._scanning_logic.depth_img_is_xz:
            self.update_roi_depth(h=h_pos)
        else:
            self.update_roi_depth(h=v_pos)
```

2. "poimangui.py"  -- roll back from line: 559
below is from the same file with a few commits roll back:
```python
    @QtCore.Slot(bool)
    def toggle_poi_selector(self, is_active):
        if is_active != self._mw.poi_selector_Action.isChecked():
            self._mw.poi_selector_Action.blockSignals(True)
            self._mw.poi_selector_Action.setChecked(is_active)
            self._mw.poi_selector_Action.blockSignals(False)
        if is_active != self.__poi_selector_active:
            if is_active:
                self.roi_image.scene().sigMouseClicked.connect(self.create_poi_from_click)
                self.roi_image.setCursor(QtCore.Qt.CrossCursor)
            else:
                self.roi_image.scene().sigMouseClicked.disconnect()
                self.roi_image.setCursor(QtCore.Qt.ArrowCursor)
        self.__poi_selector_active = is_active
        return

    @QtCore.Slot(object)
    def create_poi_from_click(self, event):
        # Only create new POI if the mouse click event has not been accepted by some other means
        # In our case this is most likely the POI marker to select the active POI from.
        if not event.accepted:
            # Z position from ROI origin, X and Y positions from click event
            new_pos = self.poimanagerlogic().roi_origin
            cursor_pos = self.roi_image.getViewBox().mapSceneToView(event.scenePos())
            new_pos[0] = cursor_pos.x()
            new_pos[1] = cursor_pos.y()
            # Check if position lies within axis boundaries.
            x_axis_box = self._mw.roi_map_ViewWidget.getAxis('bottom').sceneBoundingRect()
            y_axis_box = self._mw.roi_map_ViewWidget.getAxis('left').sceneBoundingRect()
            min_x = self.roi_image.getViewBox().mapSceneToView(y_axis_box.bottomRight()).x()
            min_y = self.roi_image.getViewBox().mapSceneToView(x_axis_box.topLeft()).y()
            if new_pos[0] <= min_x or new_pos[1] <= min_y:
                return
            # Check if position is within scan image boundaries
            image_extent = self.poimanagerlogic().roi_scan_image_extent
            if image_extent is None:
                return
            if image_extent[0][0] > new_pos[0] or image_extent[0][1] < new_pos[0]:
                return
            if image_extent[1][0] > new_pos[1] or image_extent[1][1] < new_pos[1]:
                return
            event.accept()
            self.sigAddPoiByClick.emit(new_pos)
        return
```

## Expected behavior (i.e. solution)

- [ ] just like normal

## Other Comments

- Also have a NoneTypeError (in save_logic.py), when adding POI in it's manager GUI, turns out changing codes as below can be a quic fix, but seems to change the function's behaver. 

- "logic/save_logic.py" line 568:

```python
            #param_dict = args[0].update(kwargs)
            #FIXME
            param_dict = args[0]
```

Neverhorst, 2019-05-06:
Again complete disregard of issue template and no structured description of a single problem.
Worst thing is you never describe how to replicate the issue (as the template suggests).
You also mix 3 different unrelated issues.

Nonetheless, the only issue I can reproduce is the one mentioned in "Other Comments". Will be fixed with PR #511

closed 2019-05-06 by Neverhorst

diamond2nv, 2019-05-07:
Sorry about "replicate the issue", just forgot. I was thinking about a quick solution.